open-mmlab/OpenPCDet #425: heading is strange

passion3394, 2021-01-06:
I run demo.py with my custom point cloud on KITTI pretrained model, but got the following result:

[{'pred_boxes': array([
       [ 25.383146  , -19.727966  ,  -0.542957  ,   4.5564404 ,
          1.7085134 ,   1.5574344 ,   6.1157165 ],
       [ 50.531635  ,  15.150455  ,  -0.90850633,   4.3082213 ,
          1.5774832 ,   1.5206814 ,   3.0684369 ],
       [ 37.682465  ,  16.04488   ,  -1.1022838 ,   4.3100157 ,
          1.6422797 ,   1.583584  ,   3.053904  ],
       [ 40.8249    ,  -8.533322  ,  -0.42192677,   4.378151  ,
          1.7457869 ,   1.4089336 ,   6.2229733 ],

My question is why heading in the first line is 6.1157165, in my opinion, the heading is between -np.pi to +np.pi.
But here comes the 6.1157165?

jihanyang, 2021-01-07:
Please read our README to refer our definition about the heading angle.

passion3394, 2021-01-07:
@jihanyang thanks for your reply, sorry for I can not find the definition in README.md

jihanyang, 2021-01-08:
You can refer to this figure: https://github.com/open-mmlab/OpenPCDet/blob/master/docs/dataset_vs_model.png. The direction of heading angle is from x to y in anticlockwise.

closed 2021-01-10 by jihanyang