open-mmlab/OpenPCDet #426: When I use my own datasets to train using cbgs and multihead, only the car have results

Xiangzhaohong, 2021-01-07:
CLASS_NAMES: ['vehicle', 'pedestrian', 'bicycle', 'motorcycle']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/robosense_dataset.yaml
    POINT_CLOUD_RANGE: [-51.2, -51.2, -5.0, 51.2, 51.2, 3.0] # 0.2
    #POINT_CLOUD_RANGE: [-60.0, -60.0, -5.0, 60.0, 60.0, 3.0]  # 0.15
    BALANCED_RESAMPLING: True
    DATA_PROCESSOR:
        - NAME: mask_points_and_boxes_outside_range
          REMOVE_OUTSIDE_BOXES: True

        - NAME: shuffle_points
          SHUFFLE_ENABLED: {
            'train': True,
            'test': True
          }

        - NAME: transform_points_to_voxels
          VOXEL_SIZE: [0.2, 0.2, 8.0]
          MAX_POINTS_PER_VOXEL: 20
          MAX_NUMBER_OF_VOXELS: {
            'train': 30000,
            'test': 30000
          }

MODEL:
    NAME: PointPillar

    VFE:
        NAME: PillarVFE
        WITH_DISTANCE: False
        USE_ABSLOTE_XYZ: True
        USE_NORM: True
        NUM_FILTERS: [64]

    MAP_TO_BEV:
        NAME: PointPillarScatter
        NUM_BEV_FEATURES: 64

    BACKBONE_2D:
        NAME: BaseBEVBackbone
        LAYER_NUMS: [3, 5, 5]
        LAYER_STRIDES: [2, 2, 2]
        NUM_FILTERS: [64, 128, 256]
        UPSAMPLE_STRIDES: [0.5, 1, 2]
        NUM_UPSAMPLE_FILTERS: [128, 128, 128]

    DENSE_HEAD:
        NAME: AnchorHeadMulti
        CLASS_AGNOSTIC: False

        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        USE_MULTIHEAD: True
        SEPARATE_MULTIHEAD: True
        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'vehicle',
                'anchor_sizes': [[5.0, 2.1, 1.7]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [0.0],
                'align_center': False,
                'feature_map_stride': 4,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
            },
            {
                'class_name': 'pedestrian',
                'anchor_sizes': [[0.7, 0.7, 1.7]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [0.0],
                'align_center': False,
                'feature_map_stride': 4,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.4
            },
            {
                'class_name': 'bicycle',
                'anchor_sizes': [[1.8, 0.8, 1.6]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [0.0],
                'align_center': False,
                'feature_map_stride': 4,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            },
            {
                'class_name': 'motorcycle',
                'anchor_sizes': [ [ 1.8, 0.8, 1.6 ] ],
                'anchor_rotations': [ 0, 1.57 ],
                'anchor_bottom_heights': [ 0.0 ],
                'align_center': False,
                'feature_map_stride': 4,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            }

        ]

        SHARED_CONV_NUM_FILTER: 64

        RPN_HEAD_CFGS: [
          {
            'HEAD_CLS_NAME': ['vehicle'],
          },
          {
            'HEAD_CLS_NAME': ['bicycle', 'motorcycle'],
          },
          {
            'HEAD_CLS_NAME': ['pedestrian'],
          }
        ]

        SEPARATE_REG_CONFIG:
          NUM_MIDDLE_CONV: 1
          NUM_MIDDLE_FILTER: 64
          REG_LIST: ['reg:2', 'height:1', 'size:3', 'angle:2']

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder
            BOX_CODER_CONFIG: {
              'code_size': 7,
              'encode_angle_by_sincos': True  # 为true的话会在上面那个9基础上再加1。这样就能和上面那个REG_LIST对应上
            }

        LOSS_CONFIG:
            REG_LOSS_TYPE: WeightedL1Loss
            LOSS_WEIGHTS: {
              'pos_cls_weight': 1.0,
              'neg_cls_weight': 2.0,
              'cls_weight': 1.0,
              'loc_weight': 0.25,
              'dir_weight': 0.2,
              'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: True
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.2
            NMS_PRE_MAXSIZE: 1000
            NMS_POST_MAXSIZE: 80

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 8
    NUM_EPOCHS: 40

    OPTIMIZER: adam_onecycle
    LR: 0.001
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

Xiangzhaohong, 2021-01-07:
this is my config
could u give some suggestions? thanks!
@sshaoshuai

Xiangzhaohong, 2021-01-07:
sorry, I test in ros and find it's my eval's problem, it can detect other class

closed 2021-01-07 by Xiangzhaohong

zhutongseu1994, 2021-06-29:
> CLASS_NAMES: ['vehicle', 'pedestrian', 'bicycle', 'motorcycle']
>
> DATA_CONFIG:
> _BASE_CONFIG_: cfgs/dataset_configs/robosense_dataset.yaml
> POINT_CLOUD_RANGE: [-51.2, -51.2, -5.0, 51.2, 51.2, 3.0] # [...]

请问您的训练效果怎么样 bicycle还可以嘛

zhutongseu1994, 2021-06-29:
> CLASS_NAMES: ['vehicle', 'pedestrian', 'bicycle', 'motorcycle']
>
> DATA_CONFIG:
> _BASE_CONFIG_: cfgs/dataset_configs/robosense_dataset.yaml
> POINT_CLOUD_RANGE: [-51.2, -51.2, -5.0, 51.2, 51.2, 3.0] # [...]

请问您bicycle的训练效果怎么样？
@Xiangzhaohong