open-mmlab/OpenPCDet #431: Velodyne Puck (VLP-16)

soldierofhell, 2021-01-12:
Hi, the results on VLP-16 on pretrained models aren't obviously impressive. Has anyone managed to improve them with somehow? I mean for example one can think of:
1. Retrain on dataset reduced to 16 rings
2. Aggregate point clouds (taking into account odometry) to lower frequency, but more dense data?
3. Change the position or angle of the sensor?
4. Two or more sensors? Or even fusion with depth camera?
...

Regards,

Mfertega, 2021-02-02:
Hi @soldierofhell I am looking for an application on an VLP-16 as well, have you found some good results or datasets to train?

mkolodziejczyk-piap, 2021-02-02:
So far, I've tested a little bit of 2., but the problem is that you can correct the points by movement of vehicle, but still you're left with movement of object ("blur effect"). I must do some more tests, but still I rather prefer to aggregate then reduce..

Mfertega, 2021-02-02:
I guess you need to track the movement for making those corrections, but that will not have the same result as a 64 rings lidar because the points could superpose right?.
Have you tried to reduce the number of points in the kitti dataset?.

mkolodziejczyk-piap, 2021-02-03:
> I guess you need to track the movement for making those corrections,

Yes, but that's no problem, you have to do this on real vehicle/simulator anyway

> but that will not have the same result as a 64 rings lidar because the points could superpose right?.

Of course. You'll get denser cloud but the "structure" will be different

> Have you tried to reduce the number of points in the kitti dataset?.

Not yet, but I think I'll start with Waymo

FlyerMaxwell, 2021-07-14:
> Hi, the results on VLP-16 on pretrained models aren't obviously impressive. Has anyone managed to improve them with somehow? I mean for example one can think of:
>
> 1. Retrain on dataset reduced to 16 [...]

Hi， have you solved your problem? I have met a similar problem with a lidar of 32 lines.

closed 2022-03-08 by github-actions[bot]

cennaro, 2022-03-18:
Hi, could you kindly share the evaluation metrics obtained with point clouds from VLP-16?
I am using A2D2 dataset but the results are really bad.

Car AP@0.70, 0.70, 0.70:
bbox AP:27.9918, 27.9322, 35.5970
bev  AP:19.8531, 20.3522, 26.0223
3d   AP:11.2460, 11.5596, 13.6794
aos  AP:26.38, 26.32, 33.20
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:25.8497, 25.8407, 34.3857
bev  AP:13.8069, 14.9976, 21.5022
3d   AP:3.5310, 4.0570, 7.6144
aos  AP:24.21, 24.15, 31.86
Car AP@0.70, 0.50, 0.50:
bbox AP:27.9918, 27.9322, 35.5970
bev  AP:41.1764, 40.8807, 48.6118
3d   AP:35.0433, 34.2128, 41.3431
aos  AP:26.38, 26.32, 33.20
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:25.8497, 25.8407, 34.3857
bev  AP:38.4422, 38.9257, 45.9213
3d   AP:31.6660, 31.4929, 39.0600
aos  AP:24.21, 24.15, 31.86

I would like to understand if it is all the fault of low resolution or I am doing something wrong.