open-mmlab/OpenPCDet #439: Demo on Custom data set in kitti format?

Suri12990, 2021-01-20:
Hello,
@sshaoshuai  
Firstly, I would like to thank OpenPCDet developers for their great work in Lidar perception. 

I tried the demo on Kitti data set and it worked fine. But on my data set in Kitti format, the demo is not working, I mean no prediction and bounding boxes on the objects. 
I checked other issues in this repo like [this](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md), [this](https://github.com/open-mmlab/OpenPCDet/issues/176) and [this](https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-675535776l), but I did not find solution to my problem. 

custom data will be loaded in kitti format(.bin) and can visualize in mayavi, but no prediction and bounding boxes on the objects. I also changed the Z values(z+1.6 m, optional according to Demo.md) and unified coordinates narratives and I checked the co-ordinates system in mayavi UI, still not solved.

I think demo has nothing to do with __get__item in kitti_dataset.py, since part of this code is not being executed at all, when I checked with print statement. 
What I understood from the code, I do not need to change anything in the code for demo on custom data apart from values in Z axis and unified co-ordinates system. Since data is directly feed into the model for prediction/evalution, custom data may be comply with model trained with kitti data set. 

Could anyone familiar with this help me out to solve this?

Thanks in advance

closed 2021-01-20 by Suri12990

ekorudiawan, 2021-02-04:
how do you solve this issue?

oscar-lorente, 2021-03-24:
> [image]
> Hello,
> @sshaoshuai
> Firstly, I would like to thank OpenPCDet developers for their great work in Lidar perception.
> 
> I tried the demo on Kitti data set and it worked fine. But on my data set in Kitti format, the demo is not working, I mean no prediction and bounding boxes on the objects.
> I checked other issues in this repo like [this](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md), [this](https://github.com/open-mmlab/OpenPCDet/issues/176) and [this](https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-675535776l), but I did not find solution to my problem.
> 
> custom data will be loaded in kitti format(.bin) and can visualize in mayavi, but no prediction and bounding boxes on the objects. I also changed the Z values(z+1.6 m, optional according to Demo.md) and unified coordinates narratives and I checked the co-ordinates system in mayavi UI, still not solved.
> 
> I think demo has nothing to do with __get__item in kitti_dataset.py, since part of this code is not being executed at all, when I checked with print statement.
> What I understood from the code, I do not need to change anything in the code for demo on custom data apart from values in Z axis and unified co-ordinates system. Since data is directly feed into the model for prediction/evalution, custom data may be comply with model trained with kitti data set.
> 
> Could anyone familiar with this help me out to solve this?
> Here is the output of the demo
> 
> `/home/surendra/pcdet/OpenPCDet/pcdet/models/roi_heads/pvrcnn_head.py:135: UserWarning: This overload of nonzero is deprecated: nonzero() Consider using one of the following signatures instead: nonzero(*, bool as_tuple) (Triggered internally at /pytorch/torch/csrc/utils/python_arg_parser.cpp:882.) dense_idx = faked_features.nonzero() # (N, 3) [x_idx, y_idx, z_idx] [{'pred_boxes': tensor([], device='cuda:0', size=(0, 7)), 'pred_scores': tensor([], device='cuda:0'), 'pred_labels': tensor([], device='cuda:0', dtype=torch.int64)}] `
> 
> Thanks in advance

Hey @Suri12990 , I'm facing the same problem with my custom data. Were you able to solve it? If so, how did you do it? Thanks in advance!

oscar-lorente, 2021-03-31:
Solved! The problem was related to the data format, and it worked by storing my point clouds in np.float32 (.bin files). See https://github.com/open-mmlab/OpenPCDet/issues/176#issuecomment-670702150 for more details.

HuangVictorAuto, 2021-07-20:
@Suri12990 , Hi, I faced the same problem to use a pre trained kitti model with my own data. My data can be shown in the mayavi, but with no output and prediction. I saw you close the issue, How did you tackle the problem? Thank you in advance. Danke!

[image]

mariya12290, 2021-08-06:
Hey @HuangVictorAuto 

I do not know which 3D model you are using?. If you are using Voxelnet based model, please check the position of your LiDAR from the ground. In Kitti, the position of LiDAR is 1.73 m, but the position of my LiDAR was near to the ground. So the model did not perform well on custom data. The model needs training again with the correct scene height.  If the position difference between two LiDARs is less, then the pre-trained model will perform well, else you need to train the model for custom data. 

Hope that it is useful!

Best Regards
Surendra Kumar

AOOOOOA, 2022-05-08:
> 

Hi, may I ask what is your original point cloud format? Since my original format is .bin bus also failed. I'm just curious where the problem is.

Thanks!

OrangeSodahub, 2022-07-18:
> > 
> 
> Hi, may I ask what is your original point cloud format? Since my original format is .bin bus also failed. I'm just curious where the problem is.
> 
> Thanks!

Hey, you can refer my success example using kitti format dataset and see whether it could help you : https://github.com/OrangeSodahub/CRLFnet/blob/master/src/site_model/src/LidCamFusion/OpenPCDet/pcdet/datasets/custom/README.md

OrangeSodahub, 2022-07-26:
Hello, you can refer to my successful example using kitti format custom dataset. README describes how to label, train, inference it including transformation of coordinates. It may solve your problems!
https://github.com/OrangeSodahub/CRLFnet#lid-cam-fusion
https://github.com/OrangeSodahub/CRLFnet/blob/master/src/site_model/src/LidCamFusion/OpenPCDet/pcdet/datasets/custom/README.md
You can also review this pull requrest: https://github.com/open-mmlab/OpenPCDet/pull/1032