open-mmlab/OpenPCDet #441: How to generate the results of the Waymo test set for submission?

jialeli1, 2021-01-24:
I found that there is no _test.txt_ about Waymo in the [ImageSet](https://github.com/open-mmlab/OpenPCDet/tree/master/data/waymo/ImageSets) directory.
Can someone please share instructions on how to generate the results of the Waymo test set for submission?

Thanks a lot.

jihanyang, 2021-01-26:
Hello, after downloading the test set in waymo and generating `test.txt` by yourself, you can then create infos and train the model.

jialeli1, 2021-01-26:
> Hello, after downloading the test set in waymo and generating `test.txt` by yourself, you can then create infos and train the model.

Thanks for your kind tip. I'll have a try.

jialeli1, 2021-01-26:
> Hello, after downloading the test set in waymo and generating `test.txt` by yourself, you can then create infos and train the model.

Hi, I found another problem is that there is no function for generating waymo detection result file in the code base.

In kitti_dataset.py, [generate_prediction_dicts()](https://github.com/open-mmlab/OpenPCDet/blob/a7cf5368d9cbc3969b4613c9e61ba4dcaf217517/pcdet/datasets/kitti/kitti_dataset.py#L313) saves the detection results to a file (*.txt). But in waymo_dataset.py, [generate_prediction_dicts()](https://github.com/open-mmlab/OpenPCDet/blob/a7cf5368d9cbc3969b4613c9e61ba4dcaf217517/pcdet/datasets/waymo/waymo_dataset.py#L153) does not save any files.

The [official instruction](https://github.com/waymo-research/waymo-open-dataset/blob/master/docs/quick_start.md#submit-to-leaderboard) of waymo requires the *.bin file to generate the *.txtpb file to upload to waymo server. So, could you kindly share the part of data conversion code?

HenryJunW, 2021-03-15:
Similar question, do you find a solution?

jialeli1, 2021-03-15:
Not yet.