open-mmlab/OpenPCDet #442: cida execitopm failed with error 98

alawliat, 2021-01-24:
Hi everyone,

I've been trying to get the demo to run for the past 3 days but I am getting errors when trying to run it.
I was able to build spconv1.0 successfully on python 3.6 with torch 1.1 but I can't figure this out.
Any idea what could be causing this? I have the latest NVIDIA drivers for my gtx 960m which are 460 with cuda 11.2.

I followed this exactly to build spncovn 1.0 https://github.com/traveller59/spconv/issues/152

When I try and execute: 

python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml \
    --ckpt pv_rcnn_8369.pth \
    --data_path training/velodyne/000008.bin

I get this

2021-01-24 14:19:29,135   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2021-01-24 14:19:29,179   INFO  Total number of samples: 	1
2021-01-24 14:19:30,493   INFO  ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2021-01-24 14:19:31,102   INFO  ==> Done (loaded 367/367)
2021-01-24 14:19:31,124   INFO  Visualized sample index: 	1
Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 91, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/xxx/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/xxx/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/xxx/OpenPCDet/pcdet/models/backbones_3d/spconv_backbone.py", line 139, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/xxx/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/xxx/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/spconv/modules.py", line 123, in forward
    input = module(input)
  File "/home/xxx/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/xxx/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/spconv/conv.py", line 151, in forward
    self.stride, self.padding, self.dilation, self.output_padding, self.subm, self.transposed, grid=input.grid)
  File "/home/xxx/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/spconv/ops.py", line 89, in get_indice_pairs
    stride, padding, dilation, out_padding, int(subm), int(transpose))
RuntimeError: /home/xxx/spconv_8da6f96/src/spconv/indice.cu 120
cuda execution failed with error 98

I also tried using pointrcnn and I got a different error:

error: Internal: no kernel image is available for execution on the device

MartinHahner, 2021-01-24:
Unfortunately, your GPU is too old. 
It is based on the **Maxwell** architecture, but _spconv_ only supports **Pascal** and later.
You can find out more about the CUDA architectures [here](https://en.wikipedia.org/wiki/CUDA#GPUs_supported).

alawliat, 2021-01-24:
That makes sense thank you, I have an RTX 2070 Max Q Design which is based on the Turing architecture as well, this should be okay correct?

MartinHahner, 2021-01-25:
Yes, **Turing** architecture should work.

closed 2021-01-26 by alawliat

twn29004, 2021-03-19:
Hi, can you tell me how to resolve it. I have the  same problem.

twn29004, 2021-03-19:
And my gpu is gtx 3090,I don't know which file i need to chang

Thomas-Kb, 2021-09-15:
> Unfortunately, your GPU is too old.
> It is based on the **Maxwell** architecture, but _spconv_ only supports **Pascal** (and later).
> You can find out more about the CUDA architectures [here](https://en.wikipedia.org/wiki/CUDA#GPUs_supported).
> 
> Sorry.

I trained PointPillars and PVRCNN **successfully** on a Maxwell-GPU by using the docker image for the sparse convolution library (provided by the author [here](https://hub.docker.com/r/scrin/dev-spconv)). The straightforward way would be to add the layers for OpenPCDet on top of the provided Dockerfile. Check out [this](https://yunshuang.medium.com/setting-up-running-environment-for-the-open-source-3d-object-detection-project-openpcdet-using-7427c0091edd) medium article, where the author describes a similar approach.

jayadeepk, 2022-01-02:
I solved it for older architectures (Maxwell) using this method: https://github.com/pangsu0613/CLOCs/issues/20#issuecomment-1003668924