open-mmlab/OpenPCDet #450: About results on validation in the paper

Ultra11nstinct, 2021-02-03:
Hi Shi，
Thanks again for your great work.
I just wanna know the ablation study in your PV-RCNN paper is trained for car，cyclist，pedestrian but only reported car results， or trained only for car class？

Wish every success in your work.
Looking forward to your reply.

jihanyang, 2021-02-09:
Hello,
Since the result in car is more stable, we only accomplish the ablation study on car.

Ultra11nstinct, 2021-02-10:
> Hello,
> 
> Since the result in car is more stable, we only accomplish the ablation study on car.

Thanks for your reply @jihanyang 
so,it means that in ablation study only cars are trained.
There is another question, is the different class result(cars,pedestrians,cyclists) in the partA2 paper obtained by using one model,or multi models?

closed 2022-03-09 by github-actions[bot]