open-mmlab/OpenPCDet #453: Converting to tensorrt on Xavier AGX development board

SM1991CODES, 2021-02-14:
HI,

Great work!! Thank you.
Recently, after a grueling 3 day trial, I was able to get your toolkit installed and running on the Jetson Xavier AGX kit.
However, I wish to benchmark models like PointPillars and Second using tensorrt (torch2trt). Can you please help me with some guides regarding this?

I cannot convert the model directly since it takes several inputs as a dictionary.

Best Regards
Sambit

DirectionLynn, 2021-04-12:
> HI,
>
> Great work!! Thank you.
> Recently, after a grueling 3 day trial, I was able to get your toolkit installed and running on the Jetson Xavier AGX kit.
> However, I wish to benchmark models like [...]

Duplicate of #

> HI,
>
> Great work!! Thank you.
> Recently, after a grueling 3 day trial, I was able to get your toolkit installed and running on the Jetson Xavier AGX kit.
> However, I wish to benchmark models like [...]

i can not install the toolkit with run python setup.py develop
and the error shows like this
Traceback (most recent call last):
  File "setup.py", line 106, in <module>
    'src/sampling_gpu.cu',
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/__init__.py", line 163, in setup
    return distutils.core.setup(**attrs)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/develop.py", line 38, in run
    self.install_for_development()
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/develop.py", line 140, in install_for_development
    self.run_command('build_ext')
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 87, in run
    _build_ext.run(self)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 708, in build_extensions
    build_ext.build_extensions(self)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 208, in build_extension
    _build_ext.build_extension(self, ext)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 456, in unix_wrap_single_compile
    cflags = unix_cuda_flags(cflags)
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 423, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 1561, in _get_cuda_arch_flags
    arch_list[-1] += '+PTX'
IndexError: list index out of range
do you know how to fix it?  thanks

turtleMaster20, 2021-05-27:
Using the Jetson Xavier NX and trying to do the same thing. Any Leads?

turtleMaster20, 2021-05-27:
> Duplicate of #
> 
> > HI,
> > Great work!! Thank you.
> > Recently, after a grueling 3 day trial, I was able to get your toolkit installed and running on the Jetson Xavier AGX kit.
> > However, I wish to benchmark models like PointPillars and Second using tensorrt (torch2trt). Can you please help me with some guides regarding this?
> > I cannot convert the model directly since it takes several inputs as a dictionary.
> > Best Regards
> > Sambit
> 
> i can not install the toolkit with run python setup.py develop
> and the error shows like this
> Traceback (most recent call last):
> File "setup.py", line 106, in
> 'src/sampling_gpu.cu',
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/**init**.py", line 163, in setup
> return distutils.core.setup(**attrs)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/core.py", line 148, in setup
> dist.run_commands()
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/dist.py", line 955, in run_commands
> self.run_command(cmd)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/dist.py", line 974, in run_command
> cmd_obj.run()
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/develop.py", line 38, in run
> self.install_for_development()
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/develop.py", line 140, in install_for_development
> self.run_command('build_ext')
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/cmd.py", line 313, in run_command
> self.distribution.run_command(command)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/dist.py", line 974, in run_command
> cmd_obj.run()
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 87, in run
> _build_ext.run(self)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 339, in run
> self.build_extensions()
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 708, in build_extensions
> build_ext.build_extensions(self)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
> self._build_extensions_serial()
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
> self.build_extension(ext)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 208, in build_extension
> _build_ext.build_extension(self, ext)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
> depends=ext.depends)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/distutils/ccompiler.py", line 574, in compile
> self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 456, in unix_wrap_single_compile
> cflags = unix_cuda_flags(cflags)
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 423, in unix_cuda_flags
> cflags + _get_cuda_arch_flags(cflags))
> File "/home/nvidia/miniforge-pypy3/envs/sad/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 1561, in _get_cuda_arch_flags
> arch_list[-1] += '+PTX'
> IndexError: list index out of range
> do you know how to fix it? thanks

Maybe try to see if you have all the dependencies installed? For me I was not able to install 'numba' but as I was not training in the Jetson I though it not necessary.

Can you check your Cuda and CuDnn version? Also please check that pytorch and spconv are installed and are the compatible versions.

closed 2022-03-09 by github-actions[bot]

henaDanielWu, 2024-04-15:
> Duplicate of #
>
>
> i can not install the toolkit with run python setup.py develop and the error shows like this Traceback (most recent call last): File "setup.py", line 106, in 'src/sampling_gpu.cu', [...]

Have you solved this problem? I also encountered this problem.