open-mmlab/OpenPCDet #455: Training the pointrcnn error~

laukalfp, 2021-02-23:
I trying to training the pointrcnn with kitti datasets, i got this error. The PartA2.yaml PartA_free.yaml have problem either.
python train.py --cfg_file ./cfgs/kitti_models/pointrcnn.yaml
python train.py --cfg_file ./cfgs/kitti_models/PartA2.yaml 

I can't find the "USE_ROAD_PLANE: False" in those yaml, how can i need to do?
the pv_rcnn.yaml work well with train.py, and the pointrcnn.yaml with demo.py work well.

`
2021-02-23 16:14:17,289   INFO  **********************Start training cfgs/kitti_models/pointrcnn(default)**********************
epochs:   0%|                                                                                                                                                                                                                   | 0/80 [00:00<?, ?it/s]
Traceback (most recent call last):                                                                                                                                                                                            | 0/1856 [00:00<?, ?it/s]
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/tools/train_utils/train_utils.py", line 19, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 819, in __next__
    return self._process_data(data)
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 846, in _process_data
    data.reraise()
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/_utils.py", line 369, in reraise
    raise self.exc_type(msg)
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 380, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/dataset.py", line 124, in prepare_data
    'gt_boxes_mask': gt_boxes_mask
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 93, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 197, in __call__
    data_dict = self.add_sampled_boxes_to_scene(data_dict, sampled_gt_boxes, total_valid_sampled_dict)
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 125, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']
KeyError: 'road_plane'

2021-02-23 14:40:25,474   INFO  **********************Start training cfgs/kitti_models/PartA2_free(default)**********************
epochs:   0%|                                                                                                                                                                                                                   | 0/50 [00:00<?, ?it/s]
Traceback (most recent call last):                                                                                                                                                                                            | 0/1856 [00:00<?, ?it/s]
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/tools/train_utils/train_utils.py", line 19, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 819, in __next__
    return self._process_data(data)
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 846, in _process_data
    data.reraise()
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/_utils.py", line 369, in reraise
    raise self.exc_type(msg)
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/fupeng/anaconda3/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 380, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/dataset.py", line 124, in prepare_data
    'gt_boxes_mask': gt_boxes_mask
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 93, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 197, in __call__
    data_dict = self.add_sampled_boxes_to_scene(data_dict, sampled_gt_boxes, total_valid_sampled_dict)
  File "/home/fupeng/fupeng/algorithm/detection3d/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 125, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']
KeyError: 'road_plane'`

laukalfp, 2021-02-23:
I found it, there still have another *.yaml. Now it can train well. 
it is in the kitti_dataset.yaml

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        - NAME: gt_sampling
          USE_ROAD_PLANE: False 

2021-02-23 18:39:22,635   INFO  **********************Start training cfgs/kitti_models/pointrcnn(default)**********************
epochs:   0%|                                                                                                                                                                                              | 0/80 [02:35<?, ?it/s, loss=2.23, lr=0.001]^Cain:  15%|████████████████████████████▋                                                                                                                                                             | 286/1856 [02:35<13:42,  1.91it/s, total_it=286]

closed 2021-02-23 by laukalfp

paathelb, 2021-09-06:
This worked for me! Can you explain why we have to set USE_ROAD_PLANE to False? What is the intuition behind this?

SaMeEr9597, 2022-03-12:
I have the same question, did you get to know why do we need to set USE_ROAD_PLANE to False?

paathelb, 2022-03-14:
I think it's because the points in the ground plane have to be removed.