open-mmlab/OpenPCDet #456: nuScenes eval error

Mengziyang-Doing, 2021-02-25:
Hi！

I added a front view learning module in OpenPCDet. It trains normally on the kitti dataset, but the eval error occurs on the nuScenes dataset. 
Is it caused by multi_head? Which codes do I need to modify?

Module:
   (vfe): PillarVFE(
    (pfn_layers): ModuleList(
      (0): PFNLayer(
        (linear): Linear(in_features=9, out_features=64, bias=False)
        (norm): BatchNorm1d(64, eps=0.001, momentum=0.01, affine=True, track_running_stats=True)
      )
    )
  )
  (map_to_bev_module): PointPillarScatter()
  (cfe): PillarCFE(
    (pfn_layers): ModuleList(
      (0): PFNLayer(
        (linear): Linear(in_features=9, out_features=64, bias=False)
        (norm): BatchNorm1d(64, eps=0.001, momentum=0.01, affine=True, track_running_stats=True)
      )
    )
  )
  (map_to_bev_front_module): PointPillarScatter_front()

Error:
Loading annotations for val split from nuScenes version: v1.0-trainval
100%|██████████████████████████████████████████████████████████████████████████████| 6019/6019 [00:17<00:00, 337.74it/s]
Loaded ground truth annotations for 6019 samples.
Filtering predictions
Traceback (most recent call last):
  File "train.py", line 202, in <module>
    main()
  File "train.py", line 195, in main
    dist_test=dist_train
  File "/home/zmeng/OpenPCDet/tools/test.py", line 121, in repeat_eval_ckpt
    result_dir=cur_result_dir, save_to_file=args.save_to_file
  File "/home/zmeng/OpenPCDet/tools/eval_utils/eval_utils.py", line 113, in eval_one_epoch
    output_path=final_output_dir
  File "/home/zmeng/OpenPCDet/pcdet/datasets/nuscenes/nuscenes_dataset.py", line 264, in evaluation
    verbose=True,
  File "/home/zmeng/anaconda3/envs/PCDet/lib/python3.7/site-packages/nuscenes/eval/detection/evaluate.py", line 94, in __init__
    self.pred_boxes = filter_eval_boxes(nusc, self.pred_boxes, self.cfg.class_range, verbose=verbose)
  File "/home/zmeng/anaconda3/envs/PCDet/lib/python3.7/site-packages/nuscenes/eval/common/loaders.py", line 222, in filter_eval_boxes
    box = eval_boxes.boxes[first_key][0]
IndexError: list index out of range

Thank you!

closed 2021-02-27 by Mengziyang-Doing

Javier-DlaP, 2021-03-23:
I'm having the same issue, how did you solve it?

Mengziyang-Doing, 2021-03-23:
> I'm having the same issue, how did you solve it?

My code has bugs that are not easy to find, you can also check your code carefully

Javier-DlaP, 2021-03-23:
But as I saw, the problem shows up using the --eval_all flag, evaluating only the last ckpt it works properly. So I'm going to use only the --ckpt flag. Thanks.

ahmedius2, 2021-12-02:
In the nuscenes/eval/common/loaders.py file of nuscenes-devkit==1.0.5, replacing the following part of the code block with the other one given at the bottom appears to solve the issue:
```
first_key = list(eval_boxes.boxes.keys())[0]
box = eval_boxes.boxes[first_key][0]
```
replace to:
```
sample_key = None
for key in list(eval_boxes.boxes.keys()):
  if len(eval_boxes.boxes[key]) > 0:
    sample_key=key
    break
if sample_key == None:
  return eval_boxes # Nothing to filter
box = eval_boxes.boxes[sample_key][0]
```