open-mmlab/OpenPCDet #459: Question about Kitti test split

GalaxyWM, 2021-03-01:
Thanks for ur PCDet!!!
There are some questions about submission to kitti test.
I randomly split train+val set into 80% train set and 20% val set to train PV-RCNN, but I only get the AP of 80%  on the moderate difficlty.
Are there any tricks to train the model?
thx ^▽^

closed 2021-03-14 by GalaxyWM

ZhangYu1ing, 2021-06-10:
Hi, may I ask where did you modify the percentage of train set and val set? Because I checked that in the kitti_dataset.yaml, it gives the val_set info to the test?

Thanks in advance.

[image]

1035843537, 2021-08-21:
> Hi, may I ask where did you modify the percentage of train set and val set? Because I checked that in the kitti_dataset.yaml, it gives the val_set info to the test?
> 
> Thanks in advance.
> 
> [image]

Hi,I'm confused about this question too.Have you solved it? May I ask how did you modify this percentage ?

Thank you!

bqcs, 2024-01-14:
> Hi,I'm confused about this question too.Have you solved it? May I ask how did you modify this percentage ?

hi，I've also encountered a similar problem, have you resolved it?