open-mmlab/OpenPCDet #463: How to use TensorBoard during training?

RolandoAvides, 2021-03-10:
Hi there!

How to use TensorBoard during training?

Thanks in advance!

Suri12990, 2021-03-11:
Hey @RolandoAvides 

Go to this folder   Openpcdet_Env/OpenPCDet/output/kitti_models/pv_rcnn/default and typer tensorboard --logdir tensorboard

Hope it works!

RolandoAvides, 2021-03-11:
Hi! Yes, it worked! I was missing TensorFlow in my environment.

Thanks!

closed 2021-03-11 by RolandoAvides

yaxi333, 2021-06-30:
Hi, 

I want to ask have you ever used the tensorboard to save and visualize the details of layers, weights, and biases of the model? If so, could you please give me some hints because I have stuck in how to visualize the model by tensorboard.

Thanks in advance!

MartinHahner, 2021-06-30:
@yaxi333: https://wandb.ai is better suited for that :wink:

swaggywilliam, 2023-12-15:
> @yaxi333: https://wandb.ai is better suited for that 😉

Could you please provide a tutorial on how to integrate W&B into OpenPCdet? Thank you!