open-mmlab/OpenPCDet #467: how to get the results on Kitti

ChenZhiKai66, 2021-03-17:
Hey, I have run test.py on the test set and save the required prediction results to files. But I don't know how to get the results on Kitti .
Thank you in advance for your help.

RolandoAvides, 2021-03-21:
Are you trying to check the results referring to the pre-trained model or one of your training checkpoints?

ChenZhiKai66, 2021-03-26:
> Are you trying to check the results referring to the pre-trained model or one of your training checkpoints?

Yes

RolandoAvides, 2021-03-29:
If you run test.py you should see the results on the kitti val set on your console. What are you getting on your console?

ChenZhiKai66, 2021-03-29:
> If you run test.py you should see the results on the kitti val set on your console. What are you getting on your console?

I didn't know how to get the results from the Kitti test set before, but now I understand that I have to upload my own prediction results on the website. So it's OK to test on the val set
It should be like this, right?

RolandoAvides, 2021-03-29:
Yes. The creators of kitti do not provide the test set labels. To evaluate the models performance on the test set one must send the model and code to the kitti website. OPEN-PCDet only evaluates on the val set.

closed 2021-03-29 by ChenZhiKai66

baraujo98, 2021-04-19:
Furthermore, you cannot get results on the test set if you are not publishing a paper with some novelty. So you can only use the validation set. From the KITTI website: "As more and more non-published work and re-implementations of existing work is submitted to KITTI, we have established a new policy: from now on, only submissions with significant novelty that are leading to a peer-reviewed paper in a conference or journal are allowed. Minor modifications of existing algorithms or student research projects are not allowed. Such work must be evaluated on a split of the training set."