open-mmlab/OpenPCDet #473: 求助，运行train.py有两个错误，难以解决

XiaXingLuo, 2021-03-26:
Traceback (most recent call last):                                                                                                                                                | 0/3712 [00:00<?, ?it/s]
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/xiaxingluo/PycharmProjects/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/home/xiaxingluo/PycharmProjects/OpenPCDet/tools/train_utils/train_utils.py", line 19, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/xiaxingluo/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 582, in __next__
    return self._process_next_batch(batch)
  File "/home/xiaxingluo/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 606, in _process_next_batch
    raise Exception("KeyError:" + batch.exc_msg)
Exception: KeyError:Traceback (most recent call last):
  File "/home/xiaxingluo/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 99, in _worker_loop
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/home/xiaxingluo/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 99, in <listcomp>
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/home/xiaxingluo/PycharmProjects/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 382, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/xiaxingluo/PycharmProjects/OpenPCDet/pcdet/datasets/dataset.py", line 124, in prepare_data
    'gt_boxes_mask': gt_boxes_mask
  File "/home/xiaxingluo/PycharmProjects/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 93, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/home/xiaxingluo/PycharmProjects/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 198, in __call__
    data_dict = self.add_sampled_boxes_to_scene(data_dict, sampled_gt_boxes, total_valid_sampled_dict)
  File "/home/xiaxingluo/PycharmProjects/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 125, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'],
KeyError: 'road_plane'

RolandoAvides, 2021-03-26:
If you didn't download the road plane dataset, you must change the configuration .yaml file.

There is a boolean parameter called road_plane that must be turned on or off, whether you have or not the road_plane dataset.

Hope it helps!

XiaXingLuo, 2021-03-27:
> If you didn't download the road plane dataset, you must change the configuration .yaml file.
> 
> There is a boolean parameter called road_plane that must be turned on or off, whether you have or not the road_plane dataset.
> 
> Hope it helps!

谢谢，我有roadplane数据，在planes下面，有三个文件夹，calib、gt_image_2、image_2，我调试代码，在kitti_dataset.py ,378行road_plane = self.get_road_plane(sample_idx)，获取到的数据是空，我进入函数发现，源代码在读取road_planes是这样进行的plane_file = self.root_split_path / 'planes' / ('%s.txt' % idx)，planes文件夹下是没有直接的txt文本，所以读取的roadplane为空。继续往下走，在database_sample.py124行，却需要data_dict['road_planes'],因为前面没有获取planes数据，所以datd里面是没有road_planes这个字典的。我现在不知道如何处理。

RolandoAvides, 2021-03-29:
Did you download the road_plane data set you your kitti folder?

XiaXingLuo, 2021-03-29:
> Did you download the road_plane data set you your kitti folder?

Yes,I download the road_plane data and set the data by the getting_started.md ,moreover I check the road_plane, the concrete road data as mentioned above.And I run ''python3 kitti_dataset.py create_kitti_infos'' successfully, I can also run "python3 test.py --cfg_file cfgs/kitti_models/pointrcnn.yaml --batch_size 1 --ckpt pointrcnn.pth --save_to_file", but I failed to run train.py when I set other parameters correctly.

RolandoAvides, 2021-03-29:
Try to set the configs in the pvrcnn configuration script:

```
USE_ROAD_PLANE: False
````

jihanyang, 2021-03-30:
> Try to set the configs in the pvrcnn configuration script:
> 
> ```
> USE_ROAD_PLANE: False
> ```

Use road plane or not will not cause significant difference.

XiaXingLuo, 2021-03-30:
> Try to set the configs in the pvrcnn configuration script:
> 
> ```
> USE_ROAD_PLANE: False
> ```

Yeah,thanks a lot,It can solve the problem of running pv_rcnn in this way,but I want to compare pointrcnn with pointrcnn in pcdet,the problem of running pointrcnn hasn't been solved,I want to know if there something wrong in the original code.Finally thanks for your reply sincerely again. 

> > Try to set the configs in the pvrcnn configuration script:
> > ```
> > USE_ROAD_PLANE: False
> > ```
> 
> Use road plane or not will not cause significant difference.

Thanks,It can solve the problem of running pv_rcnn,but not solve the problem of pointrcnn,there isn't 'USE_ROAD_PLANE: 'config   
in pointrcnn.yaml.I have studied the author's another code named pointrcnn,now It's said that the pointrcnn in pcdet has great improvements,so I want to test and study,but now I fail  to run "python train.py --cfg_file cfgs/kitti_models/pointrcnn.yaml'' ,the detailed error information is the same as what I ask for help.

jihanyang, 2021-03-30:
Hello @XiaXingLuo,
It seems that you cannot find the *.txt in `kitti/training/planes`? Have you already download the planes from https://drive.google.com/file/d/1d5mq0RXRnvHPVeKx6Q612z0YRO1t2wAp/view? Can you show your files structure of kitti/training as our `GETTING_STARTED.md`

XiaXingLuo, 2021-03-31:
> Hello @XiaXingLuo,
> It seems that you cannot find the *.txt in `kitti/training/planes`? Have you already download the planes from https://drive.google.com/file/d/1d5mq0RXRnvHPVeKx6Q612z0YRO1t2wAp/view? Can you show your files structure of kitti/training as our `GETTING_STARTED.md`

Thanks,It  is helpful for me.Thanks for your advice,I want to consult another question about debug code,when I debug the 'train.py',It stops at  the "dataloader_iter = iter(train_loader)",the console doesn't show any error information, just pause at "
INFO  **********************Start training kitti_models/pointrcnn(default)**********************
epochs: 0it [00:00, ?it/s]"
why I can't continue to debug,whether because the large data?I want to enter the network's forward function.

closed 2022-03-09 by github-actions[bot]

BiboyQG, 2024-03-26:
> Thanks,It is helpful for me.Thanks for your advice,I want to consult another question about debug code,when I debug the 'train.py',It stops at the "dataloader_iter = iter(train_loader)",the console [...]
I want to know why this is like this as well, since my progress bar just stuck at epoch 0 and not going forward...