open-mmlab/OpenPCDet #474: [SLURM TRAIN]

RolandoAvides, 2021-03-26:
Hi! Can anyone help me with the slurm trian.sh configuration? I am trying to train a model with 2 or 3 GPUs. 

This is the script I'm currently using for a simple test in my servers.

´´´
#!/bin/bash
#SBATCH --gres=gpu:1
# other options you might want
python3 train.py --cfg_file /home/rmoreira/OpenPCDet/tools/cfgs/kitti_models/pv_rcnn.yaml --batch_size 1 --workers 0 --epochs 80 --max_ckpt_save_num 10
# etc
´´´
However, I need to use the slurm sh of this repository.
How do I need to change the official Open-PCDet slurm_train.sh?

Thanks in advance!

jihanyang, 2021-03-27:
Hello, have you try our slurm_train.sh in tools/scripts? What's the problem of it?

RolandoAvides, 2021-03-29:
Hi! Not yet, I can try it this week (due to busy GPUs). I just thought that the shell script should be configurated according to the slurm specifications of each user.

I'll give you feedback after I run the shell script.

Thanks!

closed 2021-04-08 by RolandoAvides