open-mmlab/OpenPCDet #475: Training SECOND on the kitti

hyunjunChhoi, 2021-03-26:
Training pv-RCNN on the Kitti gets the simiilar results with the model zoo 

however, training SECOND on the Kitti get big gap with the model zoo 

What's the reason ... ? ??? 

I also get the similar results with model zoo when loading the pre trained model 

It might be the problem in the second.yaml ?? Could you check??

Thanks in advance for your great work

eval: 100%|██████| 236/236 [01:57<00:00,  2.01it/s, recall_0.3=(0, 3672) / 4357]
2021-03-27 08:16:41,102   INFO  *************** Performance of EPOCH 80 *****************
2021-03-27 08:16:41,102   INFO  Generate label finished(sec_per_example: 0.0311 second).
2021-03-27 08:16:41,103   INFO  recall_roi_0.3: 0.000000
2021-03-27 08:16:41,103   INFO  recall_rcnn_0.3: 0.849115
2021-03-27 08:16:41,103   INFO  recall_roi_0.5: 0.000000
2021-03-27 08:16:41,103   INFO  recall_rcnn_0.5: 0.794967
2021-03-27 08:16:41,103   INFO  recall_roi_0.7: 0.000000
2021-03-27 08:16:41,103   INFO  recall_rcnn_0.7: 0.589649
2021-03-27 08:16:41,105   INFO  Average predicted number of objects(3769 samples): 6.708
2021-03-27 08:16:54,408   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:85.5383, 83.0333, 81.5709
bev  AP:80.7454, 78.9053, 72.6542
3d   AP:74.1660, 65.5456, 59.7114
aos  AP:85.50, 82.59, 80.85
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:89.9710, 85.6153, 81.0622
bev  AP:82.5814, 79.8535, 75.7065
3d   AP:74.3151, 64.7246, 61.0487
aos  AP:89.92, 85.13, 80.33
Car AP@0.70, 0.50, 0.50:
bbox AP:85.5383, 83.0333, 81.5709
bev  AP:86.4790, 85.2595, 83.9319
3d   AP:86.2901, 84.7972, 83.3330
aos  AP:85.50, 82.59, 80.85
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:89.9710, 85.6153, 81.0622
bev  AP:91.0298, 89.8137, 85.1390
3d   AP:90.7889, 89.2118, 84.5580
aos  AP:89.92, 85.13, 80.33
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:48.8957, 45.8046, 41.3212
bev  AP:36.0074, 30.4039, 28.7422
3d   AP:30.5714, 25.1091, 23.2764
aos  AP:26.90, 25.02, 22.31
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:48.5846, 42.6401, 39.3000
bev  AP:33.2769, 27.5398, 24.1607
3d   AP:27.8157, 21.9467, 19.3384
aos  AP:26.22, 23.32, 21.20
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:48.8957, 45.8046, 41.3212
bev  AP:59.3867, 51.5480, 49.2743
3d   AP:58.6952, 50.7681, 48.4654
aos  AP:26.90, 25.02, 22.31
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:48.5846, 42.6401, 39.3000
bev  AP:60.2103, 50.9379, 46.3926
3d   AP:58.5373, 50.2283, 45.5860
aos  AP:26.22, 23.32, 21.20
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:40.7332, 36.9917, 32.3095
bev  AP:30.6835, 26.1260, 25.5869
3d   AP:29.2091, 24.4912, 24.2681
aos  AP:33.64, 31.45, 27.94
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:37.1584, 32.5252, 30.0954
bev  AP:30.8385, 24.8979, 23.6444
3d   AP:28.2701, 22.5000, 21.3864
aos  AP:29.09, 26.05, 24.27
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:40.7332, 36.9917, 32.3095
bev  AP:40.3858, 36.3693, 36.0714
3d   AP:40.3858, 36.3466, 36.0562
aos  AP:33.64, 31.45, 27.94
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:37.1584, 32.5252, 30.0954
bev  AP:37.6360, 31.9041, 30.4973
3d   AP:37.6360, 31.8937, 30.4830
aos  AP:29.09, 26.05, 24.27

2021-03-27 08:16:54,408   INFO  Result is save to /home/numb7315/OpenPCDet-master/output/kitti_models/second/default/eval/eval_with_train/epoch_80/val
2021-03-27 08:16:54,408   INFO  ****************Evaluation done.*****************
2021-03-27 08:16:54,425   INFO  Epoch 80 has been evaluated
Wait 30 seconds for next check (progress: 0.0 / 0 minutes): /home/numb7315/OpenP2021-03-27 08:17:24,455   INFO  **********************End evaluation kitti_models/second(default)**********************

jihanyang, 2021-03-27:
We have reproduced second several times. The result in model zoo should be reproducible. Could you provide your cfg?

hyunjunChhoi, 2021-03-27:
> We have reproduced second several times. The result in model zoo should be reproducible. Could you provide your cfg?

Thanks, I just use basic cfg that is provided

tools/cfgs/kitti_models/second.yaml

CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG: 
    _BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml

MODEL:
    NAME: SECONDNet

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5, 5]
        LAYER_STRIDES: [1, 2]
        NUM_FILTERS: [128, 256]
        UPSAMPLE_STRIDES: [1, 2]
        NUM_UPSAMPLE_FILTERS: [256, 256]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Car',
                'anchor_sizes': [[3.9, 1.6, 1.56]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-1.78],
                'align_center': False,
                'feature_map_stride': 8,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
            },
            {
                'class_name': 'Pedestrian',
                'anchor_sizes': [[0.8, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 8,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            },
            {
                'class_name': 'Cyclist',
                'anchor_sizes': [[1.76, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 8,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            }
        ]

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'dir_weight': 0.2,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: False
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.01
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.003
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

closed 2022-03-09 by github-actions[bot]