open-mmlab/OpenPCDet #476: The performance of OpenPCDet's piontpillars and SECOND

deepmeng, 2021-03-27:
Anyone can answer me about the performance of OpenPCDet's piontpillars and SECOND when compared with https://github.com/traveller59/second.pytorch and https://github.com/nutonomy/second.pytorch in the same evaluation metric. it seems that they employed different evaluation tools.

deepmeng, 2021-04-01:
Hi @sshaoshuai .  I need to confirm my comprehension on the two repos with you.
The mAP (Car@R11) of OpenPCDet's SECOND in KITTI is 78.62 and this mAP is for Moderate mode.
The mAP which was provided by SECOND（https://github.com/traveller59/second.pytorch）with (car.fhd.config + 160 epochs, 3D, Moderate) is 78.43.

It seems that the two repos' performance is very similar. Is this conclusion right?

jihanyang, 2021-04-01:
Yes.

deepmeng, 2021-04-01:
> Yes.

Many thanks！

closed 2021-04-01 by deepmeng