open-mmlab/OpenPCDet #480: need use tf tensor during load data

MengAaron, 2021-03-30:
Hi, I want to use range image converted from points, because I need some online gt augmentation, so i can't save range image in my disk. waymo_open_dataset is in tensorflow format, so I need to convert numpy to tensorflow. But i find that, if i add a single code, "tf.convert(np.array([1,2,3]))" , that useless code in waymo.py __getitem__(), do not return with any variable, when load data from cpu to gpu, CUDA error will occur.
change:
tf.convert(np.array([1,2,3])) in __getitem__()
error:
pcdet/models/__init__.py
batch_dict[key] = torch.from_numpy(val).float().cuda()
RuntimeError: CUDA error: unspecified launch failure

jihanyang, 2021-03-30:
I personally recommend you to re-write the tf `convert point cloud to range image` api of waymo with numpy.

MengAaron, 2021-03-31:
that will cost a lot of time! Do you have other solution?

jihanyang, 2021-03-31:
No. I think it is easy to implement and I have do this thing before.

MengAaron, 2021-03-31:
Thanks! But I am a green hand in this area, Can you share code with me? 

> No. I think it is easy to implement and I have do this thing before.

jihanyang, 2021-03-31:
I could put it on my gist later. However, I am busy recently and I need some time to find it, this might take several days.

MengAaron, 2021-04-02:
Excause me! Have you find your code? Plz！

MengAaron, 2021-04-02:
update:
Today, I find even data of tf format is on cpu, that will take about 130M memory on GPU. So, I seek methods to locally disable the GPU visibility of tensorflow, not global because data should finally be load on GPU, and in evaluation stage, tensorflow need GPU

jihanyang, 2021-04-03:
Sorry, I am really busy recently. I have already fount those functions, but I fail to access gist due to the network condition. You can send me a email. I will share those functions to you directly. Please cliam the copyright of those functions if you use them.

closed 2021-04-05 by MengAaron