open-mmlab/OpenPCDet #481: Segmentation fault (core dumped)

Luan-Zhaoliang, 2021-03-31:
Hi, when I run the code of the pointnet2 model, I met the problem of Segmentation fault (core dumped). Some of the issues have raised this question, but no one to answer it. Therefore, I beg you can help us to solve this problem.

jihanyang, 2021-03-31:
Please provide more information, otherwise I also cannot help you.

Luan-Zhaoliang, 2021-03-31:
Hi, I tested your pointnet2 model from [https://github.com/sshaoshuai/Pointnet2.PyTorch/tree/5a4416f51ceaeba242828cabf39133433336850d](url)
I can run it successfully on my local Linux Ubuntu20 with Pytorch3.6 and Python=3.6. However, when  I upload the full code to  the Linux based server and test my code on my own 3d LiDAR dataset (Oxford RobotCar Dataset), I got an error like this:
[image]
Besides, the configurations on the Linux server are same as my local Ubuntu20 system. Therefore, I am so confused abouut this problem.

jihanyang, 2021-03-31:
Sorry, I don't familar with that repository. It would be better to raise issue in that repository.

closed 2021-04-06 by jihanyang