open-mmlab/OpenPCDet #482: Quick Demo - mayavi and qt: qt.qpa.xcb: could not connect to display

ReunionJ, 2021-03-31:
Hello, I encountered the following problems when I ran the code:
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

I beg you can help me to solve this problem. Thank you!

RolandoAvides, 2021-03-31:
Hello,

I recently ran into this error on my own computer using the kitti_viz repository. The problem is related to mayavi. I will try to look into this error next week. Try to reinstall mayavi first.

jihanyang, 2021-03-31:
This error is caused by some conflict of the mayavi and python version. Could you provide your mayavi and python version?

Here is a possible solution: 
1. uninstall all pyqt5 packages in pip.
2. install qt and qtconsole in conda.

This solution works for me.

ReunionJ, 2021-04-02:
Thank you for your reply！
My mayavi and python version :
 mayavi-4.7.2
 python3.7

> This error is caused by some conflict of the mayavi and python version. Could you provide your mayavi and python version?
> 
> Here is a possible solution:
> 
> 1. uninstall all pyqt5 packages in pip.
> 2. install qt and qtconsole in conda.
> 
> This solution works for me.

Thank you for your reply！
My mayavi and python version :
 mayavi-4.7.2
 python3.7

ReunionJ, 2021-04-02:
> This error is caused by some conflict of the mayavi and python version. Could you provide your mayavi and python version?
> 
> Here is a possible solution:
> 
> 1. uninstall all pyqt5 packages in pip.
> 2. install qt and qtconsole in conda.
> 
> This solution works for me.

Thank you for your reply！
My mayavi and python version :
 mayavi-4.7.2
 python3.7

ReunionJ, 2021-04-03:
> Hello,
> 
> I recently ran into this error on my own computer using the kitti_viz repository. The problem is related to mayavi. I will try to look into this error next week. Try to reinstall mayavi first.

Okay, if you have any progress, please let me know, thank you very much!

closed 2022-03-09 by github-actions[bot]