open-mmlab/OpenPCDet #484: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv

MrCrazyCrab, 2021-04-02:
when i try to run train.py ,i met the problem "undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv", i have successfully install spconv and pcdet.
My environment:
ubuntu18.04/pytorch1.2/spconv1.0/python3.6.7

RolandoAvides, 2021-04-06:
Try to reinstall pytorch and spconv. Also, try to use python >= 3.8

closed 2022-03-09 by github-actions[bot]

OuYaozhong, 2024-09-02:
I also post this [here](https://github.com/open-mmlab/OpenPCDet/issues/1344#issuecomment-2325009940)

The same.

CUDA 11.8
Python 3.11
PyTorch 2.4
Ubuntu 22.04

Rebuild is useless. 
Rebuild and then the problem occur again.

Also followed [#216](https://github.com/open-mmlab/OpenPCDet/issues/216#issuecomment-669663092), useless too.

Error is shown below:
```
Traceback (most recent call last):
  File "/home/{$USER}/tmp/OpenPCDet/tools/train.py", line 7, in <module>
    from test import repeat_eval_ckpt
  File "/home/{$USER}/tmp/OpenPCDet/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/home/{$USER}/tmp/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: /home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-311-x86_64-linux-gnu.so: undefined symbol: _ZN3c106detail14torchCheckFailEPKcS2_jRKSs
ERROR conda.cli.main_run:execute(125): `conda run python /home/{$USER}/tmp/OpenPCDet/tools/train.py --cfg_file tools/cfgs/kitti_models/my_model.yaml --batch_size 16 --epochs 32` failed. (See above for error)
```

And it is a strange results that, if I try to rebuild, it will tell me that no 'CUDA_HOME' is specified.
But I have set it, because I have built it just now.

```
Traceback (most recent call last):
  File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 59, in <module>
    make_cuda_ext(
  File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 18, in make_cuda_ext
    cuda_ext = CUDAExtension(
               ^^^^^^^^^^^^^^
  File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1076, in CUDAExtension
    library_dirs += library_paths(cuda=True)
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yiuzon/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1207, in library_paths
    if (not os.path.exists(_join_cuda_home(lib_dir)) and
                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 2416, in _join_cuda_home
    raise OSError('CUDA_HOME environment variable is not set. '
OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
```

OuYaozhong, 2024-09-02:
> Try to reinstall pytorch and spconv. Also, try to use python >= 3.8
> 
> Please show your environment configurations.

I also post this [here](https://github.com/open-mmlab/OpenPCDet/issues/1344#issuecomment-2325009940)

The same.

Use the latest Python (although python 3.12 has released) is unable to solve the problem.

CUDA 11.8
Python 3.11
PyTorch 2.4
Ubuntu 22.04

Rebuild is useless. 
Rebuild and then the problem occur again.

Also followed [#216](https://github.com/open-mmlab/OpenPCDet/issues/216#issuecomment-669663092), useless too.

Error is shown below:
```
Traceback (most recent call last):
  File "/home/{$USER}/tmp/OpenPCDet/tools/train.py", line 7, in <module>
    from test import repeat_eval_ckpt
  File "/home/{$USER}/tmp/OpenPCDet/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/home/{$USER}/tmp/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: /home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-311-x86_64-linux-gnu.so: undefined symbol: _ZN3c106detail14torchCheckFailEPKcS2_jRKSs
ERROR conda.cli.main_run:execute(125): `conda run python /home/{$USER}/tmp/OpenPCDet/tools/train.py --cfg_file tools/cfgs/kitti_models/my_model.yaml --batch_size 16 --epochs 32` failed. (See above for error)
```

And it is a strange results that, if I try to rebuild, it will tell me that no 'CUDA_HOME' is specified.
But I have set it, because I have built it just now.

```
Traceback (most recent call last):
  File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 59, in <module>
    make_cuda_ext(
  File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 18, in make_cuda_ext
    cuda_ext = CUDAExtension(
               ^^^^^^^^^^^^^^
  File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1076, in CUDAExtension
    library_dirs += library_paths(cuda=True)
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yiuzon/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1207, in library_paths
    if (not os.path.exists(_join_cuda_home(lib_dir)) and
                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 2416, in _join_cuda_home
    raise OSError('CUDA_HOME environment variable is not set. '
OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
```

OuYaozhong, 2024-09-02:
> > Try to reinstall pytorch and spconv. Also, try to use python >= 3.8
> > Please show your environment configurations.
> 
> I also post this [here](https://github.com/open-mmlab/OpenPCDet/issues/1344#issuecomment-2325009940)
> 
> The same.
> 
> Use the latest Python (although python 3.12 has released) is unable to solve the problem.
> 
> CUDA 11.8 Python 3.11 PyTorch 2.4 Ubuntu 22.04
> 
> ```
> -@hostname:~/tmp/OpenPCDet$ nvcc -V
> nvcc: NVIDIA (R) Cuda compiler driver
> Copyright (c) 2005-2022 NVIDIA Corporation
> Built on Wed_Sep_21_10:33:58_PDT_2022
> Cuda compilation tools, release 11.8, V11.8.89
> Build cuda_11.8.r11.8/compiler.31833905_0
> -@ hostname:~/tmp/OpenPCDet $ conda list | grep spconv
> spconv-cu118              2.3.6                    pypi_0    pypi
> -@hostname:~/tmp/OpenPCDet$ gcc --version
> gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
> Copyright (C) 2021 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> ```
> 
> Rebuild is useless. Rebuild and then the problem occur again.
> 
> Also followed [#216](https://github.com/open-mmlab/OpenPCDet/issues/216#issuecomment-669663092), useless too.
> 
> Error is shown below:
> 
> ```
> Traceback (most recent call last):
>   File "/home/{$USER}/tmp/OpenPCDet/tools/train.py", line 7, in <module>
>     from test import repeat_eval_ckpt
>   File "/home/{$USER}/tmp/OpenPCDet/tools/test.py", line 14, in <module>
>     from eval_utils import eval_utils
>   File "/home/{$USER}/tmp/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
>     from pcdet.models import load_data_to_gpu
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
>     from .detectors import build_detector
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
>     from .detector3d_template import Detector3DTemplate
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
>     from ...ops.iou3d_nms import iou3d_nms_utils
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
>     from . import iou3d_nms_cuda
> ImportError: /home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-311-x86_64-linux-gnu.so: undefined symbol: _ZN3c106detail14torchCheckFailEPKcS2_jRKSs
> ERROR conda.cli.main_run:execute(125): `conda run python /home/{$USER}/tmp/OpenPCDet/tools/train.py --cfg_file tools/cfgs/kitti_models/my_model.yaml --batch_size 16 --epochs 32` failed. (See above for error)
> ```
> 
> And it is a strange results that, if I try to rebuild, it will tell me that no 'CUDA_HOME' is specified. But I have set it, because I have built it just now.
> 
> ```
> Traceback (most recent call last):
>   File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 59, in <module>
>     make_cuda_ext(
>   File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 18, in make_cuda_ext
>     cuda_ext = CUDAExtension(
>                ^^^^^^^^^^^^^^
>   File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1076, in CUDAExtension
>     library_dirs += library_paths(cuda=True)
>                     ^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/yiuzon/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1207, in library_paths
>     if (not os.path.exists(_join_cuda_home(lib_dir)) and
>                            ^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 2416, in _join_cuda_home
>     raise OSError('CUDA_HOME environment variable is not set. '
> OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
> ```

I have fixed this problem by using both `gcc=11.4` and `g++=11.4`

This problem maybe caused by the `gcc>11` or the version of `gcc` and `g++` is mismatch.

I finally fixed by using `g++=11.4`, because before that, my `gcc=11.4` but `g++=12`

llz666ha, 2025-03-20:
> > > Try to reinstall pytorch and spconv. Also, try to use python >= 3.8尝试重新安装 pytorch 和 spconv。另外，尝试使用 python >= 3.8
> > > Please show your environment configurations.请显示您的环境配置。
> > 
> > 
> > I also post this [here](https://github.com/open-mmlab/OpenPCDet/issues/1344#issuecomment-2325009940)我也[在这里](https://github.com/open-mmlab/OpenPCDet/issues/1344#issuecomment-2325009940)发布这个
> > The same.  一样。
> > Use the latest Python (although python 3.12 has released) is unable to solve the problem.使用最新的 Python（虽然 python 3.12 已经发布）无法解决问题。
> > CUDA 11.8 Python 3.11 PyTorch 2.4 Ubuntu 22.04CUDA 11.8、Python 3.11、PyTorch 2.4、Ubuntu 22.04
> > ```
> > -@hostname:~/tmp/OpenPCDet$ nvcc -V
> > nvcc: NVIDIA (R) Cuda compiler driver
> > Copyright (c) 2005-2022 NVIDIA Corporation
> > Built on Wed_Sep_21_10:33:58_PDT_2022
> > Cuda compilation tools, release 11.8, V11.8.89
> > Build cuda_11.8.r11.8/compiler.31833905_0
> > -@ hostname:~/tmp/OpenPCDet $ conda list | grep spconv
> > spconv-cu118              2.3.6                    pypi_0    pypi
> > -@hostname:~/tmp/OpenPCDet$ gcc --version
> > gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
> > Copyright (C) 2021 Free Software Foundation, Inc.
> > This is free software; see the source for copying conditions.  There is NO
> > warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Rebuild is useless. Rebuild and then the problem occur again.重建是无用的。重新生成，然后问题再次出现。
> > Also followed [#216](https://github.com/open-mmlab/OpenPCDet/issues/216#issuecomment-669663092), useless too.也遵循[了 #216](https://github.com/open-mmlab/OpenPCDet/issues/216#issuecomment-669663092)，也没用。
> > Error is shown below:  错误如下所示：
> > ```
> > Traceback (most recent call last):
> >   File "/home/{$USER}/tmp/OpenPCDet/tools/train.py", line 7, in <module>
> >     from test import repeat_eval_ckpt
> >   File "/home/{$USER}/tmp/OpenPCDet/tools/test.py", line 14, in <module>
> >     from eval_utils import eval_utils
> >   File "/home/{$USER}/tmp/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
> >     from pcdet.models import load_data_to_gpu
> >   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
> >     from .detectors import build_detector
> >   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
> >     from .detector3d_template import Detector3DTemplate
> >   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
> >     from ...ops.iou3d_nms import iou3d_nms_utils
> >   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
> >     from . import iou3d_nms_cuda
> > ImportError: /home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-311-x86_64-linux-gnu.so: undefined symbol: _ZN3c106detail14torchCheckFailEPKcS2_jRKSs
> > ERROR conda.cli.main_run:execute(125): `conda run python /home/{$USER}/tmp/OpenPCDet/tools/train.py --cfg_file tools/cfgs/kitti_models/my_model.yaml --batch_size 16 --epochs 32` failed. (See above for error)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > And it is a strange results that, if I try to rebuild, it will tell me that no 'CUDA_HOME' is specified. But I have set it, because I have built it just now.而一个奇怪的结果是，如果我尝试重建，它会告诉我没有指定 'CUDA_HOME' 。但我已经设定了它，因为我刚刚建造了它。
> > ```
> > Traceback (most recent call last):
[... 56 lines cut ...]
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 1002, in run_command
    cmd_obj.run()
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/setuptools/command/build_ext.py", line 99, in run
    _build_ext.run(self)
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 365, in run
    self.build_extensions()
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 843, in build_extensions
    build_ext.build_extensions(self)
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 481, in build_extensions
    self._build_extensions_serial()
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 507, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/setuptools/command/build_ext.py", line 264, in build_extension
    _build_ext.build_extension(self, ext)
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/Cython/Distutils/build_ext.py", line 135, in build_extension
    super(build_ext, self).build_extension(ext)
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 562, in build_extension
    objects = self.compiler.compile(
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 658, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 1574, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/kemove/anaconda3/envs/spconv2/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 1909, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension