open-mmlab/OpenPCDet #485: Visualization issues

liulingfeng123, 2021-04-03:
Why is the visualization result of demo.py and kitti_object_vis very different?
[image]

jihanyang, 2021-04-03:
I haven't compare the result of kitti_object_vis. I notice that the visualization result of demo.py miss some cases in the distant region. Could you check the score of those missed cases? Maybe the reason is our model filter the predicted boxes with scores less than 0.1 (i.e., `MODEL.POST_PROCESSING.SCORE_THRESH`). However, this just my guess, since I haven't compare them before.

liulingfeng123, 2021-04-03:
Thanks for your reply, I will try it. Wish you all the best.

closed 2021-04-06 by jihanyang