open-mmlab/OpenPCDet #491: Question about train configuration

RolandoAvides, 2021-04-08:
Hello there!

I trained a detection model with a batch size of 8. However, I noticed in the train report this configuration parameter batch_per_gpu=2. I trained my model with 1 GPU. Does this means the train was done with a batch size of 8? Or batch size of 2?

Thanks!

bmankirLinker, 2021-04-08:
this should answer your question
https://github.com/open-mmlab/OpenPCDet/blob/master/tools/train.py#L69

closed 2021-04-08 by RolandoAvides

thptai, 2021-04-08:
You can see the actual batch_size via the log in your terminal.
Another way to check is looking at the training progress bar. The first
line is the epoch, the second line is the number of iterations per epoch.
Number of iterations = total_sample_frame / actual_batch_size

============================================================

>> Name : Tran Huu Phuong Tai

>> H.P. : 010-9400-1803

>> E-mail: ***@***.***

>> Automation Lab. (83354), School of Information and Communication
Engineering,

>> Sungkyunkwan University, 2066, Seobu-ro, Jangan-gu, Suwon-si,
Gyeonggi-do, Korea (440-746)

>> Tel : +82-31-290-7197

============================================================