open-mmlab/OpenPCDet #492: .bin  display

doubelejjyy, 2021-04-08:
Why is the. Bin file incomplete when mayavi is displayed(the .bin was predicted)
predicted 
[image]
original
[image]

doubelejjyy, 2021-04-08:
run quick demo with PV-RCNN in kitti

thptai, 2021-04-08:
FOV_POINTS_ONLY: True (int kitti_dataset.yaml)

It is the config of the Kitti dataset, it only train with the FOV of the front camera.

doubelejjyy, 2021-04-09:
> FOV_POINTS_ONLY: True (in kitti_dataset.yaml)
> 
> It is the config of the Kitti dataset, it only train with the FOV of the front camera.

I'm running a prediction program. In the. bin file, the pretrained model is used to predict, but the whole. bin file cannot be displayed completely.

jihanyang, 2021-04-09:
Check the point cloud range that you set.

doubelejjyy, 2021-04-12:
> Check the point cloud range that you set.

I didn't set it. I ran quick demo  with PV-RCNN in kitti

jihanyang, 2021-04-12:
The demo.py use second.yaml https://github.com/open-mmlab/OpenPCDet/blob/0642cf06d0fd84f50cc4c6c01ea28edbc72ea810/tools/demo.py#L58. The point cloud range have been set in the cfg.

closed 2022-03-09 by github-actions[bot]