open-mmlab/OpenPCDet #494: Docker container

xfbs, 2021-04-09:
Just in case anyone needs this, @0xdh and I created a docker container with OpenPCDet. You can get it by running

    docker pull xfbs/openpcdet

The source files for it are available at [xfbs/openpcdet](https://github.com/xfbs/openpcdet).

Cheers!

MartinHahner, 2021-04-09:
Nice, but your link [xfbs/openpcdet](https://github.com/xfbs/openpcdet) is not reachable.
[image]

Maybe you need to change your repository permissions.

xfbs, 2021-04-09:
Oops, my bad, the link should be [xfbs/docker-openpcdet](https://github.com/xfbs/docker-openpcdet).

closed 2022-03-09 by github-actions[bot]