open-mmlab/OpenPCDet #495: Generate the data infos

751565516, 2021-04-09:
When I generate the data infos by running the following command:python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml ,I got an error:Traceback (most recent call last):
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 438, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 414, in create_kitti_infos
    kitti_infos_test = dataset.get_infos(num_workers=workers, has_label=False, count_inside_pts=False)
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 191, in get_infos
    return list(infos)
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/concurrent/futures/_base.py", line 598, in result_iterator
    yield fs.pop().result()
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/concurrent/futures/_base.py", line 428, in result
    return self.__get_result()
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 128, in process_single_scene
    image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 69, in get_image_shape
    assert img_file.exists()
AssertionError
And then I want to train pointrcnn by running the following command:python train.py --cfg_file ./cfgs/kitti_models/pointrcnn.yaml --batch_size 16 --epochs 50,I got a error:FileNotFoundError: [Errno 2] No such file or directory: '/media/a201/299D817A2D97AD94/OpenPCDet/data/kitti/kitti_dbinfos_train.pkl'.  So I want  ask you for help,please!

jihanyang, 2021-04-10:
It seems that you miss some images.

751565516, 2021-04-10:
Ok，I will check it again------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年4月10日(星期六) 中午11:01
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [open-mmlab/OpenPCDet] Generate the data infos (#495)

li734969503, 2021-11-03:
i meet the same question, i guess you miss this order  n Dataset Preparation  :Generate the data infos by running the following command
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml

closed 2022-03-09 by github-actions[bot]