open-mmlab/OpenPCDet #498: Multi-gpu training error: semaphore_tracker

ys920826, 2021-04-12:
Thank you for your contribution. When I used 4 GPUs to train: CUDA_VISIBLE_DEVICES=0,1,2,3  bash scripts/dist_train.sh 4  --cfg_file cfgs/kitti_models/pv_rcnn.yaml    --batch_size 8  --epochs 80.   I encountered the following error:
**UserWarning: semaphore_tracker: There appear to be 65 leaked semaphores to clean up at shutdown**.  After the error occurred, individual GPUs(not all 4Gpus) were 100% occupied, but the training progress bar did not move. This problem confuses me. Do you have any other good suggestions? Thx
My environment is rtx 2080ti 11G*4
pytorch1.5.1
python3.6.9(python3.8 the same error)
spconv 1.2
CUDA 10.0.1

ys920826, 2021-04-12:
1

> Thank you for your contribution. When I used 4 GPUs to train: CUDA_VISIBLE_DEVICES=0,1,2,3 bash scripts/dist_train.sh 4 --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 8 --epochs 80. I [...]

> Thank you for your contribution. When I used 4 GPUs to train: CUDA_VISIBLE_DEVICES=0,1,2,3 bash scripts/dist_train.sh 4 --cfg_file cfgs/kitti_models/pv_rcnn.yaml --batch_size 8 --epochs 80. I [...]

buaa-ys, 2021-04-20:
Setting workers to 0 ( --workers 0) is effective，the training speed is very slow

closed 2021-04-20 by buaa-ys

zhyever, 2021-05-28:
Hi! Is there any other solution? I met the same problem.