open-mmlab/OpenPCDet #500: ValueError: Cannot take a larger sample than population when 'replace=False'

751565516, 2021-04-12:
Hello, sorry to disturb you,but I need your help.When I want  train pointrcnn to run the following command:python train.py --cfg_file ./cfgs/kitti_models/pointrcnn.yaml --batch_size 1 --epochs 100       
But I got a error:
Traceback (most recent call last):                                                                                                         | 0/3712 [00:00<?, ?it/s]
  File "train.py", line 198, in <module>
    main()
  File "train.py", line 170, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/media/a201/299D817A2D97AD94/OpenPCDet/tools/train_utils/train_utils.py", line 93, in train_model
    dataloader_iter=dataloader_iter
  File "/media/a201/299D817A2D97AD94/OpenPCDet/tools/train_utils/train_utils.py", line 19, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 819, in __next__
    return self._process_data(data)
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 846, in _process_data
    data.reraise()
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/_utils.py", line 385, in reraise
    raise self.exc_type(msg)
ValueError: Caught ValueError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/a201/anaconda3/envs/pcdet/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 380, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/dataset.py", line 139, in prepare_data
    data_dict=data_dict
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 122, in forward
    data_dict = cur_processor(data_dict=data_dict)
  File "/media/a201/299D817A2D97AD94/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 103, in sample_points
    extra_choice = np.random.choice(choice, num_points - len(points), replace=False)
  File "mtrand.pyx", line 959, in numpy.random.mtrand.RandomState.choice
ValueError: Cannot take a larger sample than population when 'replace=False'

And interestingly is when I rewrite the False to True,I run the command,it training.

So I want to know how to solve this problem and will I rewrite the False to True affect the final prediction result ?

Please help me,thank you very much!

lyh-12345, 2021-07-06:
hello，I have the same problem. Have you solved it？

751565516, 2021-07-06:
把false改为true，不影响！------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年7月6日(星期二) 上午10:35
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [open-mmlab/OpenPCDet] ValueError: Cannot take a larger sample than population when &#39;replace=False&#39; (#500)

lyh-12345, 2021-07-06:
thank you ，i will try it

751565516, 2021-07-06:
不客气，加油，一起学习！------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年7月6日(星期二) 上午10:54
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [open-mmlab/OpenPCDet] ValueError: Cannot take a larger sample than population when &#39;replace=False&#39; (#500)

closed 2022-03-09 by github-actions[bot]