open-mmlab/OpenPCDet #504: ValueError: need at least one array to stack

TianchaoHuo, 2021-04-14:
Platform: 
Ubuntu16.04 
pytorch=1.3.1
cuda=10.0

hello, while trying to train the pointpillars network using custom kitti-like dataset, I encounter a problem :

2021-04-14 10:41:50,502   INFO  **********************Start logging**********************
2021-04-14 10:41:50,502   INFO  CUDA_VISIBLE_DEVICES=ALL
2021-04-14 10:41:50,502   INFO  cfg_file         ./cfgs/kitti_models/car_kitti_pp.yaml
2021-04-14 10:41:50,503   INFO  batch_size       4
2021-04-14 10:41:50,503   INFO  epochs           80
2021-04-14 10:41:50,503   INFO  workers          8
2021-04-14 10:41:50,503   INFO  extra_tag        default
2021-04-14 10:41:50,503   INFO  ckpt             None
2021-04-14 10:41:50,503   INFO  pretrained_model None
2021-04-14 10:41:50,503   INFO  launcher         none
2021-04-14 10:41:50,503   INFO  tcp_port         18888
2021-04-14 10:41:50,503   INFO  sync_bn          False
2021-04-14 10:41:50,503   INFO  fix_random_seed  False
2021-04-14 10:41:50,503   INFO  ckpt_save_interval 1
2021-04-14 10:41:50,503   INFO  local_rank       0
2021-04-14 10:41:50,503   INFO  max_ckpt_save_num 30
2021-04-14 10:41:50,503   INFO  merge_all_iters_to_one_epoch False
2021-04-14 10:41:50,503   INFO  set_cfgs         None
2021-04-14 10:41:50,503   INFO  max_waiting_mins 0
2021-04-14 10:41:50,503   INFO  start_epoch      0
2021-04-14 10:41:50,503   INFO  save_to_file     False
2021-04-14 10:41:50,503   INFO  cfg.ROOT_DIR: /home/wu/OpenPCDet
2021-04-14 10:41:50,503   INFO  cfg.LOCAL_RANK: 0
2021-04-14 10:41:50,503   INFO  cfg.CLASS_NAMES: ['Car']
2021-04-14 10:41:50,503   INFO  
cfg.DATA_CONFIG = edict()
2021-04-14 10:41:50,503   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
2021-04-14 10:41:50,504   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2021-04-14 10:41:50,504   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2021-04-14 10:41:50,504   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2021-04-14 10:41:50,504   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.16, 0.16, 4], 'MAX_POINTS_PER_VOXEL': 32, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2021-04-14 10:41:50,504   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2021-04-14 10:41:50,504   INFO  
cfg.MODEL = edict()
2021-04-14 10:41:50,504   INFO  cfg.MODEL.NAME: PointPillar
2021-04-14 10:41:50,504   INFO  
cfg.MODEL.VFE = edict()
2021-04-14 10:41:50,504   INFO  cfg.MODEL.VFE.NAME: PillarVFE
2021-04-14 10:41:50,504   INFO  cfg.MODEL.VFE.WITH_DISTANCE: False
2021-04-14 10:41:50,505   INFO  cfg.MODEL.VFE.USE_ABSLOTE_XYZ: True
2021-04-14 10:41:50,505   INFO  cfg.MODEL.VFE.USE_NORM: True
2021-04-14 10:41:50,505   INFO  cfg.MODEL.VFE.NUM_FILTERS: [64]
2021-04-14 10:41:50,505   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2021-04-14 10:41:50,505   INFO  cfg.MODEL.MAP_TO_BEV.NAME: PointPillarScatter
2021-04-14 10:41:50,505   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 64
[... 184 lines cut ...]
    data.reraise()
  File "/home/wu/anaconda3/envs/openpcdet/lib/python3.6/site-packages/torch/_utils.py", line 385, in reraise
    raise self.exc_type(msg)
ValueError: Caught ValueError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/wu/anaconda3/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/wu/anaconda3/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/wu/anaconda3/envs/openpcdet/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/wu/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 379, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/wu/OpenPCDet/pcdet/datasets/dataset.py", line 124, in prepare_data
    'gt_boxes_mask': gt_boxes_mask
  File "/home/wu/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 93, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/home/wu/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 177, in __call__
    sampled_boxes = np.stack([x['box3d_lidar'] for x in sampled_dict], axis=0).astype(np.float32)
  File "<__array_function__ internals>", line 6, in stack
  File "/home/wu/anaconda3/envs/openpcdet/lib/python3.6/site-packages/numpy/core/shape_base.py", line 423, in stack
    raise ValueError('need at least one array to stack')
ValueError: need at least one array to stack

any ideas can be provided to solve it? Thanks in advance!

TianchaoHuo, 2021-04-14:
I have  executed the command:
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml 

and have generated following files:
[image]

TianchaoHuo, 2021-04-14:
I fixed it. I notice that there are two lines in above :

2021-04-14 10:41:50,512 INFO Database filter by min points Car: 135 => 134
2021-04-14 10:41:50,513 INFO Database filter by difficulty Car: 134 => 0

 I think there are no data from database since they are all filtered by difficulty. Therefore， I just comment the filter_by_difficulty: [-1] of the config file. Fortunately, it works ! But could anyone tell me what does so-called "difficulty"  here mean ?  

Thanks a lot~

MartinHahner, 2021-04-14:
You can read about it [here](http://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d), just search for "difficulties".

TianchaoHuo, 2021-04-15:
> You can read about it [here](http://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d), just search for "difficulties".

Thanks mate! Have a good day ^-^.

closed 2021-04-15 by TianchaoHuo