open-mmlab/OpenPCDet #505: Question about the KITTI train/val split

SUZhaoyu, 2021-04-14:
Hi, thanks for the great work! I'd want to know the models you provide in the model zoo are trained only with the _train_ split of KITTI or the 80% _train + val_ split of KITTI? Thanks for the reply.

jihanyang, 2021-04-15:
Yes

SUZhaoyu, 2021-04-15:
That means trained with 3712 samples right?

closed 2021-04-23 by SUZhaoyu

jwma0725, 2022-01-07:
Hi, I also meet the same the problem. How should I modify the configuration file-kitti_dataset.yaml?
[image]

MartinHahner, 2022-01-08:
You need to create your own custom ImageSet lists.
The original ones are in here: 
https://github.com/open-mmlab/OpenPCDet/tree/master/data/kitti/ImageSets

Then you need to create new pkl files and use them in your yaml config file.
https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md#kitti-dataset

Petros626, 2024-09-30:
@SUZhaoyu @jwma0725 When you take a look at the three files at ImageSets, you can see the numbers of entries and in the `kitti_dataset.yaml` there is only train and val used for the training.