open-mmlab/OpenPCDet #506: RuntimeError: invalid argument 0: Sizes of tensors must match except in dimension 1. Got 188 and 185 in dimension 3 at /pytorch/aten/src/THC/generic/THCTensorMath.cu:71

TianchaoHuo, 2021-04-14:
Hello, mate

could you tell me if we can train kitti-like dataset with 360 degree data? Since as long as  changing the   POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1] of config file,  I encounter an error,  no matter which number I changed :

RuntimeError: invalid argument 0: Sizes of tensors must match except in dimension 1. Got 188 and 185 in dimension 3 at /pytorch/aten/src/THC/generic/THCTensorMath.cu:71

So, could you tell me how to train with 360 degree point cloud, or with a custom point cloud range. Thanks in advance!

MartinHahner, 2021-04-14:
Maybe this [comment](https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-677473408) and thread can help you.

TianchaoHuo, 2021-04-15:
> Maybe this [comment](https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-677473408) and thread can help you.

Thanks friend, I found a solution from [here](https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-679190936). It works! thank you for guiding me to that discussion.

closed 2021-04-15 by TianchaoHuo