open-mmlab/OpenPCDet #507: Why hasn't this repo been updated for a long time?

poodarchu, 2021-04-15:
Are there any plans for new features/roadmap?

baraujo98, 2021-04-19:
I guess they are focusing on MMDetection3D, which is a broader framework than OpenPCDet

sshaoshuai, 2021-04-20:
@baraujo98 
No. These two repos are maintained by different teams. 
As mentioned in the README, we would like to keep OpenPCDet as a "clear, simple, self-contained open source project" for 3D detection. So it will focus on implementing the key modules of various 3D detection methods, instead of supporting many works with tiny variance. 

You could see that OpenPCDet is much simpler and clear than other codebases, but based on our architecture, it should also be easy to extend to more works that are supported by other codebases.

By the way, we are also busy on our own researches and do not have so much time on supporting more features, but we will try to keep it up-to-date for some existing works.

closed 2021-04-20 by sshaoshuai