open-mmlab/OpenPCDet #508: Error in KITTI dataset prepration

arxidinakbar, 2021-04-16:
I've successfully installed the required packages, but still get this error when running KITTI dataset prepration code. 
Anyone had met this issue before?? plz help
```
Traceback (most recent call last):
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/workspace/aixiding/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 438, in <module>
    save_path=ROOT_DIR / 'data' / 'kitti'
  File "/workspace/aixiding/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 414, in create_kitti_infos
    kitti_infos_test = dataset.get_infos(num_workers=workers, has_label=False, count_inside_pts=False)
  File "/workspace/aixiding/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 191, in get_infos
    return list(infos)
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/concurrent/futures/_base.py", line 586, in result_iterator
    yield fs.pop().result()
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/concurrent/futures/_base.py", line 425, in result
    return self.__get_result()
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/workspace/aixiding/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 128, in process_single_scene
    image_info = {'image_idx': sample_idx, 'image_shape': self.get_image_shape(sample_idx)}
  File "/workspace/aixiding/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 69, in get_image_shape
    assert img_file.exists()
AssertionError
```

closed 2021-04-17 by arxidinakbar

SaMeEr9597, 2022-03-10:
I'm getting the same error, could you make any progress on this? 
Thanks

MartinHahner, 2022-03-10:
It means your KITTI dataset is not complete and is missing an image the data loader is trying to load.

shkim-emily, 2024-08-09:
I'm getting the same error too... How to fix the problem?

Petros626, 2024-10-08:
> It means your KITTI dataset is not complete and is missing an image the data loader is trying to load.

yes exactly, thats the reason @shkim-emily , @arxidinakbar