open-mmlab/OpenPCDet #511: Question about real-time usage in ROS (Robot Operating System): Is there a way to circumvent the saving and loading of the point-cloud data and 'directly' (meaning no saving) prepare and input an incoming point cloud?

0xdh, 2021-04-17:
Hi, awesome repo :)

I am currently trying to use this in a ROS application, where I get a point cloud every 100 ms in a form of a `sensor_msgs.msgs.pointcloud2` and then convert that into a numpy array points and save it in a numpy-file. Then I essentially apply the logic of demo.py 

```
points = ros_numpy.numpify(pointcloud2data)
np.save('/path/to/temp.npy', points)

model = build_network(model_cfg=cfg.MODEL, num_class=len(cfg.CLASS_NAMES), dataset=demo_dataset)
model.load_params_from_file(filename='PartA2_free_7872.pth', logger=logger, to_cpu=True)
model.cuda()
model.eval()

demo_dataset = DemoDataset(
        dataset_cfg=cfg.DATA_CONFIG, 
        class_names=cfg.CLASS_NAMES,
        training=False, 
        root_path=Path('/path/to/temp.npy'), 
        ext='.npy',
        logger=logger)

with torch.no_grad():
            for idx, data_dict in enumerate(demo_dataset):
                logger.info(f'After enumeration: \t{idx + 1}')
                data_dict = demo_dataset.collate_batch([data_dict])
                print("\collated_data_dict:\n{}".format(data_dict))
                load_data_to_gpu(data_dict)
                #print("\n I changed ")
                logger.info(f'After collating batch to GPU // Before model.forward: \t{idx + 1}')
                prediction_begin = rospy.Time.now()
                pred_dicts, _ = model.forward(data_dict)
```
This saving operation however costs 4 ms (which is valuable time in this real-time application).

My question: Is there a direct way (meaning no np.save operation) to impute `points` into either the DemoDataset or can we discard DemoDataset entirely and prepare the data ourselves on the spot, e.g. with the point_feature_encoder and then forward it to the model?

Or is this simply not feasible? Do we have to save it in a file? And why?

Thanks in advance!

ozzCb, 2021-04-29:
use 'point_cloud = np.frombuffer(msg.data, dtype=np.float32).reshape(-1, num_point_channels)`
then you can use 'data_dict = demo_dataset.prepare_data(data_dict=input_dict)' to get data dict

closed 2022-03-09 by github-actions[bot]

baberjunaid, 2023-06-12:
> use `point_cloud = np.frombuffer(msg.data, dtype=np.float32).reshape(-1, num_point_channels)` then you can use `data_dict = demo_dataset.prepare_data(data_dict=input_dict)` to get data dict

I have the same issue. Once, data_dict is created. how to predict? following steps after data_dict is generated always give errors

```

load_data_to_gpu(data_dict)
pred_dicts, _ = model.forward(data_dict)
```