open-mmlab/OpenPCDet #512: RuntimeError: CUDA error: an illegal memory access was encountered

GabrieleGalimberti-GaleSelector, 2021-04-18:
Hi, when I try demo.py in tools with a custom dataset I found this error after ~10/20 pointcloud.
I'm not sure that is the size of pcd that is too big.

```
(25816, 4)
2021-04-18 16:54:53,469   INFO  Visualized sample index: 	14
Traceback (most recent call last):
  File "3Dobjectdetection_masterdegree.py", line 120, in <module>
    main()
  File "3Dobjectdetection_masterdegree.py", line 98, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/galimberti/projects/OpenPCDet/pcdet/models/detectors/PartA2_net.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/galimberti/projects/OpenPCDet/pcdet/models/roi_heads/partA2_head.py", line 200, in forward
    x_part = self.conv_part(part_features)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/spconv/modules.py", line 134, in forward
    input = module(input)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/spconv/modules.py", line 134, in forward
    input = module(input)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/spconv/conv.py", line 181, in forward
    use_hash=self.use_hash)
  File "/home/galimberti/.conda/envs/galimberti/lib/python3.6/site-packages/spconv/ops.py", line 95, in get_indice_pairs
    int(use_hash))
RuntimeError: CUDA error: an illegal memory access was encountered
```

Thanks for watching

MartinHahner, 2021-05-06:
It's the same issue as here: https://github.com/open-mmlab/OpenPCDet/issues/338
There is no solution yet.

GabrieleGalimberti-GaleSelector, 2021-05-06:
Thank you @MartinHahner 

For other viewers:
if you try to use it and you find this error, you can check if the file is created in the output directory and pass beyond i input point clouds.
Whit this method I obtain every result. 

But I don't understand from which reference system are bounding box results.
Can you help me?

HuangVictorAuto, 2021-07-19:
Hey, Gabriele, I encountered the same problem when I try to test the pre-trained model on my own data. The demo works fine with the Kitti data, but fails with my own data. Have you solved the issue?

GabrieleGalimberti-GaleSelector, 2021-07-19:
I solve partially @HuangVictorAuto  running multiple times the same program chechink if a pcd is processed. I skip pcd processed.

I lost my configuration of packages:
with config are you using for openpcdet?

HuangVictorAuto, 2021-07-20:
my configuration are as followed:
python 3.7,torch 1.7,cuda 11.0, cudnn 8.1, spconv 1.2.1,pcdet 0.3.0

closed 2022-03-09 by github-actions[bot]