open-mmlab/OpenPCDet #516: How to test all points in one frame point cloud?

Yupeng-mango, 2021-04-22:
I have used the second model to test kitti data, however there is a problem I can't solve.
Although I changed the FOV_POINTS_ONLY to False in kitti_dataset.yaml, and I recreated the kitti data files, after I run test.py to predict the result, there is only the detection results of the front field of view.
I want to get test results for all fields of view.
Could you please give me some help?
Thanks!

jihanyang, 2021-04-22:
Please change the `POINT_CLOUD_RANGE`

Yupeng-mango, 2021-05-04:
> Please change the `POINT_CLOUD_RANGE`

I make it! Thank you~

feigezuishuai123, 2021-05-04:
I want to ask what does the six values in POINT_CLOUD_RANGE mean?

jihanyang, 2021-05-04:
[min_x, min_y, min_z, max_x, max_y, max_z]

closed 2021-05-04 by jihanyang