open-mmlab/OpenPCDet #519: What's the meaning of "return partial(self.random_world_flip, config=config)" when data_dict=None?

zheyuanWang, 2021-04-23:
Hi,

I can't understand the return cases in the file */mnt/hard-disk-data/wangzheyuan/OpenPCDet_Rimg/pcdet/datasets/augmentor/data_augmentor.py*:

The data augmentation functions are called by the forward function:
```python
def forward(self, data_dict):
        """
        Args:
            data_dict:
                points: (N, 3 + C_in)
                gt_boxes: optional, (N, 7) [x, y, z, dx, dy, dz, heading]
                gt_names: optional, (N), string
                ...
        Returns:
        """
        points = data_dict['points']
        ori_shape = points.shape
        if len(ori_shape) == 3:
            data_dict['points'] = points.reshape((-1, ori_shape[-1]))
        for cur_augmentor in self.data_augmentor_queue:
            data_dict = cur_augmentor(data_dict=data_dict)
```

In the second line of one augment function, itself is returned with `return partial(self.random_world_flip, config=config)`.  Could you kindly explain why would we return like this? What's the meaning in doing so?
```python
def random_world_flip(self, data_dict=None, config=None):
        if data_dict is None:
            return partial(self.random_world_flip, config=config)
        gt_boxes, points = data_dict['gt_boxes'], data_dict['points']
        for cur_axis in config['ALONG_AXIS_LIST']:
            assert cur_axis in ['x', 'y']
            gt_boxes, points = getattr(augmentor_utils, 'random_flip_along_%s' % cur_axis)(
                gt_boxes, points,
            )

        data_dict['gt_boxes'] = gt_boxes
        data_dict['points'] = points
        return data_dict
```

MartinHahner, 2021-04-23:
The way I understand it, it "chains" multiple augmentations together (`random_world_flip` is just one of them).

For an explanation of 
```python
partial()
``` 
see [here](https://docs.python.org/3/library/functools.html#functools.partial).

zheyuanWang, 2021-04-25:
> The way I understand it, it "chains" multiple augmentations together (`random_world_flip` is just one of them).
 
Thanks for your kindly reply. 

I am wondering why would the function return itself, when no input  is given (`data_dict is None`)
```python
def random_world_flip(self, data_dict=None, config=None):
        if data_dict is None:
            return partial(self.random_world_flip, config=config)
```
From my point of view, this function does nothing when `data_dict is None`. So why bother to return such a function, whose *data_dict* & *config* are still None?

jihanyang, 2021-04-26:
This `if` condition is called in `__init__`. This can generate a function queue which is called in `forward`.

zheyuanWang, 2021-04-26:
> This `if` condition is called in `__init__`. This can generate a function queue which is called in `forward`.
Thanks for your answer!

I quote the `__init__` function here for other readers:
(called at the line `            cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)`

It gives the `cur_cfg` into the function. Partial is used to "fix" this augmentor for later usages.

```python
class DataAugmentor(object):
    def __init__(self, root_path, augmentor_configs, class_names, logger=None, **data_attr):
        self.root_path = root_path
        self.class_names = class_names
        self.logger = logger
        self.data_attr = data_attr

        self.data_augmentor_queue = []
        aug_config_list = augmentor_configs if isinstance(augmentor_configs, list) \
            else augmentor_configs.AUG_CONFIG_LIST

        for cur_cfg in aug_config_list:
            if not isinstance(augmentor_configs, list):
                if cur_cfg.NAME in augmentor_configs.DISABLE_AUG_LIST:
                    continue
            cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
            self.data_augmentor_queue.append(cur_augmentor)
```

closed 2021-04-26 by zheyuanWang