open-mmlab/OpenPCDet #520: "dataloader_iter = iter(train_loader)" is too slow

ca3907, 2021-04-27:
Hi,
I am currently training on multiple GPUs.

During training, I noticed that there was an excessive time delay between the epochs.

I seem to be spending more than 30 seconds in this phrase.
`dataloader_iter = iter(train_loader)`

Is there any solution?

MartinHahner, 2021-04-27:
I also observe this, I'm curious if there is a "solution", too.

ca3907, 2021-04-28:
> I also observe this, I'm curious if there is a "solution", too.

There was no problem When I using nn.DataParallel, not DistributedDataParallel
So I thought there was a solution.
Thanks for your answer!

Pyten, 2021-04-28:
I had met the same problem. Even worse, when I expand my dataset size, training stacks using DistributedDataParallel with 4 gpus. Only set workers = 0 the ddp can proceed, but the speed can't overtake training with one gpu without DDP. If anyone has some suggestions, please let me know.

closed 2022-03-09 by github-actions[bot]

chenrui17, 2022-11-11:
I had met the same problem when i use ddp with multigpu trainning

hanin97, 2024-09-12:
I had met the same problemI had met the same problem, please reopen.

Dis-Able, 2025-03-12:
I even had met the same problem, reopen plz!!!