open-mmlab/OpenPCDet #521: meaning of SAMPLE_GROUPS parameter for ground truth sampling's config file

zheyuanWang, 2021-04-28:
Hi, 

does anyone know what's the meaning of `SAMPLE_GROUPS` in the config file waymo.yaml?
```
DATA_AUGMENTOR:
    DISABLE_AUG_LIST: [ ]
    AUG_CONFIG_LIST:
        - NAME: gt_sampling_T
          USE_ROAD_PLANE: False
          DB_INFO_PATH: [gt_datainfo]
          PREPARE: {
              filter_by_min_points: [ 'Car:10', 'Pedestrian:5', 'Cyclist:5', 'Van:20' ],
              filter_by_ranges: [-50.0, -50.0, -2, 50.0, 50.0, 4]
          }
          SAMPLE_GROUPS: [ 'Car:40','Pedestrian:30', 'Cyclist:30', 'Van:10' ]
          NUM_POINT_FEATURES: 4
          DATABASE_WITH_FAKELIDAR: False
          REMOVE_EXTRA_WIDTH: [ 0.2, 0.2, 0.2 ]
          LIMIT_WHOLE_SCENE: True

```

It's set to `          SAMPLE_GROUPS: [ 'Car:40','Pedestrian:30', 'Cyclist:30', 'Van:10' ]` in my config but I am not sure about the meaning.

zheyuanWang, 2021-04-28:
I guess it's the target number of objects in each frame, judging from this line from `database_sampler`:
```python
sample_group['sample_num'] = str(int(self.sample_class_num[class_name]) - num_gt) 
```

closed 2021-04-28 by zheyuanWang

deepmeng, 2021-11-13:
It seems that sample_group parameter was set to balance the sample numbers for different categories and to eliminate the long tail.  Right?

MartinHahner, 2021-11-25:
It is there to balance the imbalance of foreground (belonging to objects we want to predict) vs. background points (not belonging to any object). Without augmenting additional objects, this imbalance would be much bigger. Still, after this augmentation, there are typically way more background than foreground points in a scene.