open-mmlab/OpenPCDet #528: Customized dataset

beedrill, 2021-05-02:
Is there a tutorial that one can follow to make a customized dataset? Thing like how to organize the data, in terms of points and labels and how to apply it?

paland3, 2021-06-16:
You can simulate KITTI dataset, i.e. place your point cloud files, images, labels in the same structure, and make your own dataset config based on kitti_dataset.yaml.

MartinHahner, 2021-06-30:
more info is given here https://github.com/open-mmlab/OpenPCDet/issues/198

closed 2022-03-09 by github-actions[bot]

OrangeSodahub, 2022-07-26:
Hello, you can refer to my successful example using kitti format custom dataset. README describes how to label, train, inference it including transformation of coordinates. It may solve your problems!
https://github.com/OrangeSodahub/CRLFnet#lid-cam-fusion
https://github.com/OrangeSodahub/CRLFnet/blob/master/src/site_model/src/LidCamFusion/OpenPCDet/pcdet/datasets/custom/README.md

OrangeSodahub, 2022-07-26:
You can also refer to this pull requrest: #1032