open-mmlab/OpenPCDet #531: About road plane

ReunionJ, 2021-05-06:
Excuse me, what do the four data in the 'road plane' file represent? It doesn't look like a、b、c、d in the formula ax+by+cz+d=0. I want to know how to get the formula of pavement according to the data in road plane. Thank you very much!
[image]
[image]
[image]

Kemo-Huang, 2021-06-30:
Further, I also want to know how to obtain the road plane data for other benchmarks on KITTI (e.g. tracking). Thanks!

a954217436, 2021-09-01:
Same questions... Do you get the answer?

BAI-Yeqi, 2022-02-25:
@ReunionJ Hi, may I know if you have figured this out? Thanks.

closed 2022-05-16 by github-actions[bot]