open-mmlab/OpenPCDet #533: error from spconv

wanglijia666, 2021-05-07:
Environment:
• Linux (tested on 16.04)
• Python 3.6.13
• PyTorch 1.4.0 
• CUDA 10.1 

when install spconv v1.2, after running python setup.py bdist_wheel, error:
/usr/local/cuda-10.1/lib64/libcublas.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
src/cuhash/CMakeFiles/cuhash.dir/build.make:206: recipe for target '../lib.linux-x86_64-3.6/spconv/libcuhash.so' failed
make[2]: *** [../lib.linux-x86_64-3.6/spconv/libcuhash.so] Error 1
CMakeFiles/Makefile2:153: recipe for target 'src/cuhash/CMakeFiles/cuhash.dir/all' failed
make[1]: *** [src/cuhash/CMakeFiles/cuhash.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

How to solve this problem, please？

MartinHahner, 2021-05-07:
Please always include the full traceback.
Presumably nobody is able to help you with such a short error description.

wanglijia666, 2021-05-08:
Sorry, here are the full details:
`
$ cat /usr/local/cuda/version.txt
CUDA Version 10.1.243

$ cat /usr/local/cuda/include/cudnn.h | grep CUDNN_MAJOR -A 2
#define CUDNN_MAJOR 7
#define CUDNN_MINOR 6
#define CUDNN_PATCHLEVEL 5
--
#define CUDNN_VERSION (CUDNN_MAJOR * 1000 + CUDNN_MINOR * 100 + CUDNN_PATCHLEVEL)

#include "driver_types.h"

$ cmake --version
cmake version 3.20.1

$ cat /proc/driver/nvidia/version
NVRM version: NVIDIA UNIX x86_64 Kernel Module  430.50  Thu Sep  5 22:36:31 CDT 2019
GCC version:  gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)

$ python setup.py bdist_wheel
running bdist_wheel
running build
running build_py
running build_ext
Release
|||||CMAKE ARGS||||| ['-DCMAKE_PREFIX_PATH=/home/wlj/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch', '-DPYBIND11_PYTHON_VERSION=3.6', '-Dexpr" -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/wlj/anaconda3/envs/pcdet/spconv/build/
-- Caffe2: CUDA detected: 10.1
-- Caffe2: CUDA nvcc is: /usr/local/cuda-10.1/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda-10.1
-- Caffe2: Header version is: 10.1
-- Found cuDNN: v7.6.5  (include: /usr/local/cuda-10.1/include, library: /usr/local/cuda-10.1/lib64/libcudnn.so)
-- Autodetected CUDA architecture(s):  7.5 7.5 7.5 7.5
-- Added CUDA NVCC flags for: -gencode;arch=compute_75,code=sm_75
CMake Warning (dev) at /opt/cmake-3.20.1/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (torch) does
  not match the name of the calling package (Torch).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /home/wlj/anaconda3/envs/pcdet/lib/python3.6/site-packages/torch/share/cmake/Torch/TorchConfig.cmake:122 (find_package_handle_standard_args)
  CMakeLists.txt:25 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Autodetected CUDA architecture(s):  7.5 7.5 7.5 7.5
-- pybind11 v2.6.3 dev1
-- Configuring done
CMake Warning (dev) in src/cuhash/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "cuhash".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/cuhash/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "cuhash".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/spconv/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
[... 53 lines cut ...]
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 299, in run
    self.run_command('build')
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 48, in run
    self.build_extension(ext)
  File "setup.py", line 92, in build_extension
    subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
  File "/home/wlj/anaconda3/envs/pcdet/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.`

wanglijia666, 2021-05-08:
I have read this solution: https://github.com/traveller59/spconv/issues/44, and I tried to modify in the following waysetup.py
Before:
```python
subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env)
subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
```
After:
```python
subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env)

build_make_file = 'build/temp.linux-x86_64-3.7/src/cuhash/CMakeFiles/cuhash.dir/build.make'
link_file = 'build/temp.linux-x86_64-3.7/src/cuhash/CMakeFiles/cuhash.dir/link.txt'

for file in [build_make_file, link_file]:
    with open(file) as f:
        newText = f.read().replace('/usr/local/cuda', env['CUDA_ROOT'])
    with open(file, "w") as f:
        f.write(newText)

for file in [build_make_file, link_file]:
            with open(file) as f:
                newText = f.read().replace('/usr/local/cuda-10.1/lib64/libcublas.so', '/usr/lib/x86_64-linux-gnu/libcublas.so')
            with open(file, "w") as f:
                f.write(newText)

subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
```
However, the same error still occur, then, how should I do to solve this question, please?

wanglijia666, 2021-05-08:
How to solve this question?
```python
/usr/local/cuda-10.1/lib64/libcublas.so: file not recognized: File format not recognized
```
$ cd /usr/local/cuda-10.1/lib64/
$ ls
[image]
$ file libcublas.so
libcublas.so: data

wanglijia666, 2021-05-10:
I install the spconv successfully by rebuilding the environment, it may be the cause of cmake, I think.

MartinHahner, 2021-05-10:
Can you run this without any errors?
`python -c "import spconv"`
It is to check if your spconv was successfully installed

wanglijia666, 2021-05-10:
```python
$ python
import spconv
```
[image]

Manueljohnson063, 2021-06-24:
@wanglijia666 
Could you please mention the cmake version you installed

wanglijia666, 2021-06-24:
> @wanglijia666
> Could you please mention the cmake version you installed

cmake 3.13.5

closed 2022-03-08 by github-actions[bot]