open-mmlab/OpenPCDet #539: Integration of PointPainting

MartinHahner, 2021-05-15:
@sshaoshuai @jihanyang 
I think [those guys](https://github.com/Song-Jingyu/PointPainting) did a great job implementing [PointPainting](https://arxiv.org/abs/1911.10150) based on your code-base. 
Maybe you can make some suggestions how to best integrate it into OpenPCDet directly.

For more info, see here:
https://github.com/Song-Jingyu/PointPainting/issues/4

sshaoshuai, 2021-05-18:
@MartinHahner , thank you for the suggestion, Martin! 
Pointpainting is an interesting work, and it should be easy to support it in OpenPCDet if we consider the image segmentation models as an external library instead of integrating it to OpenPCDet. 

I will give some suggestions to the authors at  https://github.com/Song-Jingyu/PointPainting/issues/4.

MartinHahner, 2021-06-30:
@sshaoshuai: I guess you probably have seen [PointAugmenting](https://openaccess.thecvf.com/content/CVPR2021/html/Wang_PointAugmenting_Cross-Modal_Augmentation_for_3D_Object_Detection_CVPR_2021_paper.html) by now, too, right?

It would be awesome if these types of works (**PointPainting** and **PointAugmenting**) could be incorporated into `OpenPCDet`.
I assume many methods will follow along these lines of work to fuse LiDAR and camera data.
It seems to be the best (/first) strategy to successfully merge these two modalities and get better results than LiDAR-only methods.

closed 2022-03-08 by github-actions[bot]