open-mmlab/OpenPCDet #540: Unable to install OpenPCDet

zzmonlyyou, 2021-05-16:
Ubuntu 20.04
CUDA 11.3

I installed the spconv successfully and everything works fine until python3 setup.py develop.
Here is the output after I entered python3 setup.py develop in the terminal:
/usr/local/lib/python3.8/dist-packages/torch/cuda/__init__.py:52: UserWarning: CUDA initialization: Unexpected error from cudaGetDeviceCount(). Did you run some cuda functions before calling NumCudaDevices() that might have already set an error? Error 804: forward compatibility was attempted on non supported HW (Triggered internally at  /pytorch/c10/cuda/CUDAFunctions.cpp:109.)
  return torch._C._cuda_getDeviceCount() > 0
No CUDA runtime is found, using CUDA_HOME='/usr'
running develop
running egg_info
creating pcdet.egg-info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
writing manifest file 'pcdet.egg-info/SOURCES.txt'
/usr/local/lib/python3.8/dist-packages/torch/utils/cpp_extension.py:369: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
creating build
creating build/temp.linux-x86_64-3.8
creating build/temp.linux-x86_64-3.8/pcdet
creating build/temp.linux-x86_64-3.8/pcdet/ops
creating build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms
creating build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/local/lib/python3.8/dist-packages/torch/include -I/usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include -I/usr/local/lib/python3.8/dist-packages/torch/include/TH -I/usr/local/lib/python3.8/dist-packages/torch/include/THC -I/usr/include/python3.8 -c pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o build/temp.linux-x86_64-3.8/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE="_gcc" -DPYBIND11_STDLIB="_libstdcpp" -DPYBIND11_BUILD_ABI="_cxxabi1011" -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
In file included from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Parallel.h:140,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/utils.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:5,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/nn.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/all.h:13,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/extension.h:4,
                 from pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:10:
/usr/local/lib/python3.8/dist-packages/torch/include/ATen/ParallelOpenMP.h:83: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
   83 | #pragma omp parallel for if ((end - begin) >= grain_size)
      | 
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp: In function ‘int boxes_iou_bev_cpu(at::Tensor, at::Tensor, at::Tensor)’:
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:242:55: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
  242 |     const float *boxes_a = boxes_a_tensor.data<float>();
      |                                                       ^
In file included from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Tensor.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Context.h:4,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/ATen.h:9,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:9:
/usr/local/lib/python3.8/dist-packages/torch/include/ATen/core/TensorBody.h:398:7: note: declared here
  398 |   T * data() const {
      |       ^~~~
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:243:55: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
  243 |     const float *boxes_b = boxes_b_tensor.data<float>();
      |                                                       ^
In file included from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Tensor.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/Context.h:4,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/ATen/ATen.h:9,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:9:
/usr/local/lib/python3.8/dist-packages/torch/include/ATen/core/TensorBody.h:398:7: note: declared here
  398 |   T * data() const {
      |       ^~~~
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:244:49: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
  244 |     float *ans_iou = ans_iou_tensor.data<float>();
[... 357 lines cut ...]
  File "/usr/lib/python3/dist-packages/setuptools/command/build_ext.py", line 87, in run
    _build_ext.run(self)
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/cpp_extension.py", line 708, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/lib/python3/dist-packages/setuptools/command/build_ext.py", line 208, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/usr/lib/python3.8/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/cpp_extension.py", line 456, in unix_wrap_single_compile
    cflags = unix_cuda_flags(cflags)
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/cpp_extension.py", line 423, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/cpp_extension.py", line 1561, in _get_cuda_arch_flags
    arch_list[-1] += '+PTX'
IndexError: list index out of range

Please help! Thank you so much!

MartinHahner, 2021-05-17:
Click this [link](https://letmegooglethat.com/?q=Error+804%3A+forward+compatibility+was+attempted+on+non+supported+HW) :wink: 

First search result: https://github.com/pytorch/pytorch/issues/40671

Manueljohnson063, 2021-05-18:
@MartinHahner 
Hai Sir,
I had trained my own  dataset (from velodyne 32 lidar) with Rcnn but getting wrong results ,It is producing lots of BBoxes could you please help......
Thanks in advance

turtleMaster20, 2021-05-28:
> @MartinHahner
> Hai Sir,
> I had trained my own dataset (from velodyne 32 lidar) with Rcnn but getting wrong results ,It is producing lots of BBoxes could you please help......
> Thanks in advance

Maybe you can try to remove the boxes with low scores first? This worked (to a certain extend) with pointrcnn for me. Using the pre-trained weights.

Manueljohnson063, 2021-05-28:
Thanks isir,
How did you setup your dataset ,In my case  i am only using velodyne data so i put dummy images .
Could you please give me some elaboration  
Thanks in advance
If you dont mind could you please gave me your mail ID sir....

turtleMaster20, 2021-06-02:
> @turtleMaster20
> Thanks sir,
> How did you setup your dataset ,In my case i am only using velodyne data so i put dummy images .
> Could you please give me some elaboration
> Thanks in advance
> If you dont mind could you please gave me your mail ID sir....

I used the pre-trained weights from the package. My Application is with ROS so I am using the velodyne data directly with the script from [here](https://github.com/open-mmlab/OpenPCDet/pull/193). Look for `remove_low_score_nu`.

This system architecture is completely pointcloud based so I don't understand why you have to use dummy images (or are you referring to the velodyne pointcloud as images?). 

To my understanding PointRCNN is based on PointClouds and RCNN is based on 2D Camera Images.
If you use the above mentioned script then you can just subscribe to the topic from velodyne and should work smoothly with the pre-trained weights or with your own trained weights.

KangChou, 2021-06-24:
[image]

darkdevahm, 2021-10-17:
Did you solve this issue?

SpaceView, 2021-12-02:
Just for reference, it may help

Error: List Index out of range

How to solve?

This means your relevant software package installation has some problem,  for me, it is related to nvidia, please refer to the following link for details.

https://spacevision.blog.csdn.net/article/details/121672106

Make sure your nvidia driver and nvidia prob are installed correctly.

closed 2022-03-08 by github-actions[bot]