open-mmlab/OpenPCDet #544: latest kornia caused some trouble

MartinHahner, 2021-05-21:
In both of the following settings, I had to use an old version (_v0.2_) of kornia and install it via `pip install kornia==0.2`.

Running `pip install -r requirements.txt` 
I think always wants to install the latest **kornia** (_currently 0.5.2_) which in turn wants the latest **pytorch** (_currently 1.8.1_).

local
```bash
➜ conda list | grep cud
cudatoolkit               9.2.148              h80a95b2_6    conda-forge
cudatoolkit-dev           9.2                      py36_2    conda-forge
cudnn                     7.6.5                 cuda9.2_0  
pytorch                   1.1.0           cuda92py36h65efead_0  
torchvision               0.3.0           cuda92py36h0601742_0  
```

server
```bash
- boost/1.73.0 
- cmake/3.16.5  
- cuda/10.1.243  
- cudnn/7.6.4
- gcc/6.3.0
- nccl/2.7.8-1
- openblas/0.2.20
- python_gpu/3.8.5

❯ pip list | grep torch
torch                           1.4.0
torchvision                     0.5.0
```

shijianjian, 2021-05-22:
Hi @MartinHahner, since Kornia is a functional lib that used latest Pytorch functions, we cannot fully be fully compatible with previous versions. Basically, we are supporting torch versions after 1.6.0.

/cc @edgarriba @ducha-aiki

ducha-aiki, 2021-05-22:
You can fix that via --no-deps flag. I believe, it works for the most of functions

MartinHahner, 2021-05-25:
> Hi @MartinHahner, since Kornia is a functional lib that used latest Pytorch functions, we cannot be fully compatible with previous torch versions. Basically, we are supporting torch versions after 1.6.0.
> 
> /cc @edgarriba @ducha-aiki

Yes, I understand. It is just that upgrading pytorch to 1.6.0 or higher breaks other things for me.
The reason I posted this issue is to help others that might run into the same problem.

closed 2021-05-25 by MartinHahner