open-mmlab/OpenPCDet #547: Rotation Angle

zzmonlyyou, 2021-05-26:
Hi,

I have successfully implemented the repository and thank you for your contribution. I am now using the results for object detection [x, y, z, w, l, h, Rotation angle] to graph the bounding box in the bird-eye view. Right now, I simply rotate the bounding box around the center by the rotation angle in degree, but it is not the same as the demo (which presents the correct bbox).  Thank you for your help. 
[image]

closed 2022-03-08 by github-actions[bot]

st235, 2023-11-16:
@zzmonlyyou any chance you were testing on `waymo`?

Petros626, 2024-12-09:
@st235 @zzmonlyyou guys haha what's up? did you succed with your goals?