open-mmlab/OpenPCDet #553: import iou3d_nms_cuda  ImportError: libtorch_cuda_cu.so

ddhxm, 2021-06-07:
I was tryting to do some single detect in ros,but when i run the code,i met the error.But i have succeed in installing the PCdet.Can anyone help me??Thanks a lot!
from pcdet.datasets import DatasetTemplate
  File "/home/ddh/catkin_ws_fixed/src/OpenPCDet/pcdet/datasets/__init__.py", line 7, in <module>
    from .dataset import DatasetTemplate
  File "/home/ddh/catkin_ws_fixed/src/OpenPCDet/pcdet/datasets/dataset.py", line 8, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/ddh/catkin_ws_fixed/src/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 6, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/ddh/catkin_ws_fixed/src/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 5, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/ddh/catkin_ws_fixed/src/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: libtorch_cuda_cu.so: cannot open shared object file: No such file or directory

greyweb, 2021-08-24:
Facing the same Error.

ZJDWKTX, 2021-08-30:
same error

ZJDWKTX, 2021-08-31:
@ddhxm @greyweb same error ，do you solve it？

ddhxm, 2021-09-01:
> @ddhxm @greyweb same error ，do you solve it？

hello,actually when i trying to find the problem,  i found that when i install the project locally ,and use the pip requirement, it automaticlly change the version of the torch with cuda10.2,but i had install the torch with cuda11.1,and everytime i reinstalled the project, it changed the version again.so i canceled the lines which related to torch and torchversion in  requirement.txt，and the problem solved. In my opinion it might be the version diference. You may be check the version you installed.

jihanyang, 2021-09-02:
@ddhxm I guess it is `kornia` that change your torch environment. You can try to remove it on requirement.txt and comment https://github.com/open-mmlab/OpenPCDet/blob/c9d31d393acaae34d74cb03bd6ccff9976d3d1f3/setup.py#L65 in `setup.py` if you don't aim at using CaDDN.

closed 2022-03-08 by github-actions[bot]