open-mmlab/OpenPCDet #562: TypeError: __init__() got an unexpected keyword argument 'backbone_channels'

LiZengsong, 2021-06-12:
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path /home/tong/app/OpenPCDet/object/training/velodyne/000000.bin

--ckpt pv_rcnn_8369.pth --data_path data/kitti/training/velodyne/000008.bin
2021-06-10 21:26:52,585   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2021-06-10 21:26:52,843   INFO  Total number of samples: 	1
Traceback (most recent call last):
  File "demo.py", line 103, in <module>
    main()
  File "demo.py", line 82, in main
    model = build_network(model_cfg=cfg.MODEL, num_class=len(cfg.CLASS_NAMES), dataset=demo_dataset)
  File "/home/lzs/point/pcdet/OpenPCDet/pcdet/models/__init__.py", line 18, in build_network
    model_cfg=model_cfg, num_class=num_class, dataset=dataset
  File "/home/lzs/point/pcdet/OpenPCDet/pcdet/models/detectors/__init__.py", line 26, in build_detector
    model_cfg=model_cfg, num_class=num_class, dataset=dataset
  File "/home/lzs/point/pcdet/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 7, in __init__
    self.module_list = self.build_networks()
  File "/home/lzs/point/pcdet/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 46, in build_networks
    model_info_dict=model_info_dict
  File "/home/lzs/point/pcdet/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 166, in build_roi_head
    num_class=self.num_class if not self.model_cfg.ROI_HEAD.CLASS_AGNOSTIC else 1,
TypeError: __init__() got an unexpected keyword argument 'backbone_channels'.

any ideas can be provided to solve it? Thanks in advance!

zituka, 2021-06-15:
i have the same question

MartinHahner, 2021-06-30:
Should be fixed by bugfix https://github.com/open-mmlab/OpenPCDet/commit/26a16123cf820e29ed565f4ad53964c66f3ea030, no?

closed 2022-03-08 by github-actions[bot]