open-mmlab/OpenPCDet #563: The problem when i

czy-0326, 2021-06-13:


czy-0326, 2021-06-13:
When I use demo.py to visualize the nuscenes dataset with the pretrained model(second multihead ),I load 10 sweeps,but the detection result is so terrible, there are many cars in red circles,but it can not be detected,while there are no cars in green bbox.
[image]

Can you help me deal with this problem.
Thankyou for your reply.

MartinHahner, 2021-06-30:
Some of your axis seem off. KITTI has a different coordinate system compared to nuScenes.
I think [demo.py](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/demo.py) is only meant to work with KITTI, otherwise, you maybe have to adapt the script.

closed 2022-03-08 by github-actions[bot]

LuizGuzzo, 2024-05-10:
what file did you use to play in demo.py? I've already tried some .bin files and they all have problems.