open-mmlab/OpenPCDet #570: demo.py error on jetson xavier

reddittoob, 2021-06-18:
Hi, I'm trying to run demo.py on nvidia jetson xavier board. 

I installed all packages and libraries well. 

However, when I run demo.py, I got an error like below.

2021-06-18 17:23:23.414 (  35.171s) [        7EB21010]vtkXOpenGLRenderWindow.:650    ERR| vtkXOpenGLRenderWindow (0x17b07610): Cannot create GLX context.  Aborting.

Anyone somebody know solution about this problem?

Thank you.

jihanyang, 2021-06-19:
Just don't run `mlab.show(stop=True)` and the window will not show up.

reddittoob, 2021-06-20:
I didn't run mlab.show(stop=True). Howerver, the window still show up like below image.
[image]

I don't know how to do it.

closed 2022-03-08 by github-actions[bot]

SmBito, 2022-04-16:
@reddittoob Hello, I have the same problem, have you found a solution? Thanks